adafruit/circuitpython #7109: `coproc.halt` frequently crashes

jepler, 2022-10-23:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.3-11-gedce717cfc-dirty on 2022-10-23; ESP32-S3-EYE with ESP32S3
```

### Code/REPL

```python
# Full code at https://github.com/jepler/ulp-circuitpython
import time
import struct
from coproc import *
from minidump import e, h, en

def storele32(cm, off, x):
    print(off, ":=", x)
    cm[off:off+4] = struct.pack("<l", x)
def storele16(cm, off, x):
    print(off, ":=", x)
    cm[off:off+2] = struct.pack("<h", x)
binary = e._readat(h.p_offset, h.p_filesz)
cm = CoprocMemory(0x50000000, 8176)
off = en.st_value
print(f"value at {off}, adjusted offset is {off}")
c = Coproc(binary)
try:
    run(c)

    while True:
        cm[off] = 10
        time.sleep(2)
        cm[off] = 20
        time.sleep(2)
finally:
    print("about to halt")
    time.sleep(.2)
    halt(c)
    print("returned from halt")
    time.sleep(.2)
```

### Behavior

The device disconnects from USB when the `coproc.halt()` is called.
```
>>> import do_coproc
@0000: Load 302 bytes starting at 4096
6f 00 e0 01 13 00 00 00 13 00 00 00 13 00 00 00
shared_mem @ 0x0130 0x0400 bytes
value at 304, adjusted offset is 304
<---- press ctrl-c here
about to halt

[tio 12:17:26] Disconnected
```

jepler, 2022-11-18:
As I re-test today I'm getting different results.  Of course, I'm also using different boards and different builds of CircuitPython.  Notably, these boards are non-PSRAM boards, but at least they're in the feather form factor and have the LED on GPIO13. The build is 8.0.0-beta.4-25-g195ad4e479 which is the same as b8a2d3ffdcbd396a6f551a15abfc8ca7705e8d2e except for the addition of these boards.

Here's a full example with all necessary files:
[coproc-example.zip](https://github.com/adafruit/circuitpython/files/10043120/coproc-example.zip)

The main file is called "do_coproc.py"; you can rename it to "code.py" or "import do_coproc" to run the test.

On esp32-s2 (adafruit feather esp32s2 reverse tft prototype), CircuitPython never freezes. However, after a few runs the LED stops blinking. This condition persists until the module is reset. Just hitting ctrl-c and re-starting with `import do_coproc` does not work after the LED has frozen.

on esp32-s3 (adafruit feather esp32s3 reverse tft prototype), many runs completed successfully. After hitting ctrl-c and re-starting `import do_coproc` in a fresh repl, it continues working.

of course, the LEDs blink at different rates on the two boards when using the same source program, because the "cycles per microsecond" figure is different.

dhalbert, 2022-12-10:
Moving this back to 8.0.0. I think it's going to get fixed by @tannewt's work?

dhalbert, 2022-12-21:
@tannewt Is this fixed (or is now moot) by #7218?

tannewt, 2022-12-22:
I never tested with this repro code so I'm not sure. I didn't see any issues with my (different) tests.

dhalbert, 2022-12-22:
@jepler I will close this since the underlying modules are so different now. If you'd like to retest with a new trial, that'd be great, and we can start a new issue if necessary.

closed 2022-12-22 by dhalbert