adafruit/circuitpython #7111: Unwanted CPY statusbar output in between REPL output of a running script

PaulskPt, 2022-10-24:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.3-11-gedce717cf on 2022-10-23; FeatherS3 with ESP32S3
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2022 Paulus Schulinck
#
# SPDX-License-Identifier: MIT
##############################

"""
    This CircuitPython script is created to show how I managed to
    sort received WiFi Access Points in one of the following sorting orders:
    1) on ssid;
    2) on channel;
    3) on rssi.
    See function 'pr_scanned_ap()'.

    This script also contains the function 'ck_double()' that I created to prevent listing of 'double Access Points'.
    I experienced that the list of AP's contained AP's with the same SSID, on the same CHANNEL
    but with a slight different RSSI value.
    In the function ck_doubles() AP's with a same SSID on the same CHANNEL within +/- 5 points of RSSI difference
    will be regarded as 'double' and will not be added to the list of found AP's.

    Note: the CircuitPython (V8.x) status_bar is set ON/off in file: boot.py

    This script is tested successfully on an Unexpected Maker ProS3 and FeatherS3.

    Want to see more of my work: Github @PaulskPt

"""
import time
import sys
import board
import wifi
import ipaddress
import socketpool
from collections import OrderedDict

my_debug = False  # Set to True if you need debug output to REPL
use_ping = True

start = True
ap_dict = {} # dictionary of WiFi access points
ap_cnt = 0
ip = None
s_ip = None
le_s_ip = 0
tz_offset = 0

pool = None
	
id = board.board_id

print()  # Cause REPL output be on a line under the status_bar
print(f"WiFi test for board: \'{id}\'") 
print("waiting 5 seconds...")
time.sleep(5)

if my_debug:
    if wifi is not None:
        print(f"wifi= {type(wifi)}")

try:
    from secrets import secrets
except ImportError:
[... 397 lines cut ...]
There is no need for more detailed description. See the REPL output under 'Behavior' and you can see what happens.

Note: 
This script depends on the following files:
feathers3.py
boot.py
secrets.py

### Additional information

What I think should not be is:
CircuitPython statusbar lines in between the REPL output of the running script.

This never saw this happen before. I don't like this to happen.
I think: during the execution of a script one should only see the REPL output of that script.
After ending the script it is OK to see the CPY statusbar again.

I had this 'experience' also the past days on other boards running CircuitPython V8..0.0-beta.3.
At first I didn't want to report this 'phenomena'. Instead I used a file ```boot.py``` to 
```
import supervisor
supervisor.status_bar.console = False
```

However, I think such a drastic measure shouldn't be necessary.

dhalbert, 2022-10-30:
Is this in Mu or in another terminal program?

PaulskPt, 2022-10-30:
It happens mainly in Mu:

[image]

In PuTTY the CPY V8 statusbar text is seen instead of 'PuTTY' in the caption of the PuTTY app window only
but not in the print output:

[image]

dhalbert, 2022-11-02:
The released version of Mu has no support for the escape codess produced by the status bar, and will interleave them with the REPL output. There is a fix in Mu for this, but there has been no release yet that includes it. I am inquiring about when another beta of Mu might be released.

PaulskPt, 2022-11-02:
Thank you for your response and inquiry Dan.

DJDevon3, 2023-04-28:
@PaulskPt Time to revisit for 8.0.5 stable? At least with my Mu it no longer behaves in this manner on any of my Circuit Python devices. I'm currently testing the UM FeatherS2 & S3 in Mu serial. I'm not testing with web workflow though.

dhalbert, 2023-11-07:
Closing because i think this has been addressed upstream.

closed 2023-11-07 by dhalbert