adafruit/circuitpython #7121: Circuitpython README.rst  link to readthedocs (section Documentation) redirects to Error 403 forbidden page

PaulskPt, 2022-10-25:
### CircuitPython version

```python
Current
```

### Code/REPL

```python
not relevant
```

### Behavior

[image]

### Description

On the README page ```https://github.com/adafruit/circuitpython```, README.rst, under ```Documentation```,  I clicked on the link ```Read The Docs``` (http://circuitpython.readthedocs.io/en/latest/?) . I 'landed' on an error page saying ```403 Forbidden```. (see the annotated screenshot above).

### Additional information

None.

PaulskPt, 2022-10-25:
The problem seems to be inside the https://docs.circuitpython.org/en/latest/shared-bindings/index.html.
If I enter a search term it redirects me to 403 forbidden. When I click on on of the items listed on the page, e.g. ```adapter```, ```BluetoothError``` or ```RoleError```, the same happens.

jepler, 2022-10-25:
I can't reproduce here.  I visited https://docs.circuitpython.org/en/latest/README.html then entered RoleError and selected the first/only hit. This took me to https://docs.circuitpython.org/en/latest/shared-bindings/_bleio/index.html#bleio.RoleError which displays in my browser 

If I click the link RoleError (not using search) then it also works, linking me to the same page and anchor https://docs.circuitpython.org/en/latest/shared-bindings/_bleio/index.html#bleio.RoleError

My browser is Firefox 105.0.3 on Debian Linux. Besides on my home internet in the midwest US, I also tried `curl -I` from DigitalOcean droplets in their NYC and SFO datacenters and got "HTTP/2 200" responses, which indicate success.

[image]
[image]

jepler, 2022-10-25:
It doesn't match the symptoms you reported, but it's notable that cloudflare recorded an outage today. The incident report states that there were "[Increased HTTP 530 Errors](https://www.cloudflarestatus.com/incidents/kdpqngcbbn25)" and was resolved earlier today.

PaulskPt, 2022-10-25:
Hi Jeff, thank you! You're right. I just checked, now that link works. I just saw on the Portuguese TV that there was here also an outage of WhatsApp. So I close this issue.

closed 2022-10-25 by PaulskPt