adafruit/circuitpython #7127: QT Py ESP32-S2 Wifi Unknown failure 15 on beta 3

DJDevon3, 2022-10-26:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.3 on 2022-10-20; Adafruit QT Py ESP32S2 with ESP32S2
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2022 DJDevon3 for Adafruit Industries
# SPDX-License-Identifier: MIT
# UM ESP32 FeatherS3

import wifi
import socketpool

# Initialize WiFi Pool
pool = socketpool.SocketPool(wifi.radio)

try:
    from secrets import secrets
    timezone = secrets['timezone']
    tz_offset_seconds = int(secrets['timezone_offset'])
except ImportError:
    print("Secrets File Import Error")
    raise

# Connect to Wi-Fi
print("\n===============================")
print("Connecting to WiFi...")
while not wifi.radio.ipv4_address:
    try:
        wifi.radio.connect(secrets['ssid'], secrets['password'])
    except ConnectionError as e:
        print("Connection Error:", e)
        print("Retrying in 10 seconds")

# With the 205 Error it never reaches below this point
print("Connected!\n")
```

### Behavior

Should connect to wifi. Throws error 15 instead. I should mention this happens while connected via USB. I'm not using web workflow.

Connecting to WiFi...
Connection Error: Unknown failure 15
Retrying in 10 seconds
Connection Error: Unknown failure 15
Retrying in 10 seconds

### Description

This issue does NOT affect a QT Py S3 running the same code, only affects the QT Py S2. These are the only 2 boards I've tried this code on. No idea if other S2 boards are affected by this beta 3 bug yet. 

### Additional information

Bug found by Tekktrik, I'm only confirming bug exists in my tests on same hardware.

anecdata, 2022-10-26:
`15` is `WIFI_RADIO_ERROR_4WAY_HANDSHAKE_TIMEOUT`. Is this the onboard antenna? What's the physical environment between the QT Py and the Access point?

tekktrik, 2022-10-26:
I've been in the same room as my router or in the other room when this happened, nothing too wild.

anecdata, 2022-10-26:
I can't reproduce it on either model of QT Py S2, using the above code or forever-looping connection [code](https://gist.github.com/anecdata/613f9de22ae246e59ca581fbad7de749). In the forever-looping code, there are several errors of various types among 500 good connections, but retry recovers the connection.

DJDevon3, 2022-10-26:
I'm 10 ft from my AP and the Qt Py S3 connect fine. Qt Py S2 fails every time. Cannot get a connection. I can't throw all my other boards that have no problem connecting into the mix because they're still on beta 2 or lower.

anecdata, 2022-10-26:
I don't think I've updated the bootloader on any of these boards. The one on the testbench now is: `TinyUF2 Bootloader f12dd0f - tinyusb (0.10.1-109-g9b3ec69b)`. This is one of the earliest units with onboard antenna I could get my hands on.

My first inclinations would be hardware issue or interference of some kind, though your proximity and the regularity of the issue don't reek strongly of interference.

it's possible something is getting stuck in the internals. You could try escalating recovery mechanisms if repeated retries aren't working... in this order:
```py
wifi.radio.enabled = False  # then True, then re-connect
wifi.radio.stop_station()  # then wifi.radio.start_station, then re-connect
supervisor.reload()
microcontroller.reset()
```

DJDevon3, 2022-10-26:
```py
TinyUF2 Bootloader f12dd0f - tinyusb (0.10.1-109-g9b3ec69b)
Model: Adafruit QT Py ESP32-S2
Board-ID: ESP32S2-QTPY-revB
Date: Dec 12 2021
```

anecdata, 2022-10-26:
Which [rev](https://www.adafruit.com/product/5325) board, B or C? I have two Rev B, and 10 Rev C with u.FL, in operation and don't see connection issues on any of them.

tekktrik, 2022-10-26:
Both of the ones I've noticed it on are Rev C

anecdata, 2022-10-26:
Script linked above (https://gist.github.com/anecdata/613f9de22ae246e59ca581fbad7de749). I was about to say your Board-ID says "rev B" in the comment above, but I seem to recall some discussion on Discord that those didn't necessarily match the silk. And lo and behold, they don't.

DJDevon3, 2022-10-26:
Mine has no mention of a rev on the silkscreen.

tekktrik, 2022-10-26:
Ah, mine says C but the bootloader shows B in `INFO_UF2.TXT` for both.

DJDevon3, 2022-10-26:
Well I was hoping anecdata's would actually be the different one to explain the difference, now we have a hodge podge of no consequence.

anecdata, 2022-10-26:
It should have a Ⓑ or ⓒ, not to be confused with a Copyright ;-)

DJDevon3, 2022-10-26:
Updated bootloader to
```py
TinyUF2 Bootloader 0.11.0 - tinyusb (0.12.0-203-ga4cfd1c69)
Model: Adafruit QT Py ESP32-S2
Board-ID: ESP32S2-QTPY-revB
Date: Oct 17 2022
```
It showed up with QTPYBOOT which was a neat difference. Also shows an Adafruit logo in device manager, very neat.

Loaded beta 3 on it
```py
Adafruit CircuitPython 8.0.0-beta.3 on 2022-10-20; Adafruit QT Py ESP32S2 with ESP32S2
Board ID:adafruit_qtpy_esp32s2
UID:85FC97BAF1C5
```
Ran anecdata's regression tester code, still nothing.
```py
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
]0;🐍Wi-Fi: off | code.py | 8.0.0-beta.3\]0;�Wi-Fi: No IP | code.py | 8.0.0-beta.3\Connecting... Disconnecting... 
]0;🐍Wi-Fi: off | code.py | 8.0.0-beta.3\Connecting... ]0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.3\Disconnecting... 
]0;🐍Wi-Fi: off | code.py | 8.0.0-beta.3\Connecting... ]0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.3\Disconnecting... 
]0;�Wi-Fi: off | code.py | 8.0.0-beta.3\Connecting... ]0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.3\Disconnecting... 
]0;�Wi-Fi: off | code.py | 8.0.0-beta.3\Connecting... ]0;�Wi-Fi: No IP | code.py | 8.0.0-beta.3\Disconnecting... 
]0;🐍Wi-Fi: off | code.py | 8.0.0-beta.3\Connecting... ]0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.3\Disconnecting... 
]0;�Wi-Fi: off | code.py | 8.0.0-beta.3\Connecting... ]0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.3\Disconnecting... 
]0;�Wi-Fi: off | code.py | 8.0.0-beta.3\Connecting... ]0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.3\Disconnecting... 
]0;🐍Wi-Fi: off | code.py | 8.0.0-beta.3\Connecting... ]0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.3\Disconnecting... 
]0;�Wi-Fi: off | code.py | 8.0.0-beta.3\Connecting... ]0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.3\Disconnecting... 
]0;🐍Wi-Fi: off | code.py | 8.0.0-beta.3\Connecting... ]0;�Wi-Fi: No IP | code.py | 8.0.0-beta.3\Disconnecting... 
]0;�Wi-Fi: off | code.py | 8.0.0-beta.3\Connecting... ]0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.3\Disconnecting... 
]0;�Wi-Fi: off | code.py | 8.0.0-beta.3\Connecting... ]0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.3\Disconnecting... 
]0;�Wi-Fi: off | code.py | 8.0.0-beta.3\Connecting... ]0;🐍Wi-Fi: No IP | code.py | 8.0.0-beta.3\Disconnecting... 
]0;🐍Wi-Fi: off | code.py | 8.0.0-beta.3\Connecting... ]0;�Wi-Fi: No IP | code.py | 8.0.0-beta.3\Disconnecting... 
```

chadlung, 2022-10-27:
Tested on a brand new QT PY ESP32-S2 uFL (Adafruit 5348) and attached antenna. Circuit Python Beta 3.

```
TinyUF2 Bootloader f12dd0f - tinyusb (0.10.1-109-g9b3ec69b)
Model: Adafruit QT Py ESP32-S2
Board-ID: ESP32S2-QTPY-revB
Date: Dec 12 2021
```
It is working for me.
```
Connecting... 192.168.1.27 Disconnecting... 
Connecting... 192.168.1.27 Disconnecting... 
```

DJDevon3, 2022-10-28:
@anecdata Maybe I've misunderstood what the script is supposed to show?  I'd like to think it would show an IP if it's actually connected. REPL doesn't show that, no errors. Just connect and disconnect. 

```py
import time, wifi, socketpool
from secrets import secrets

SLEEP = 1

pool = socketpool.SocketPool(wifi.radio)

def connect():
    try:
        wifi.radio.connect(secrets['ssid'], secrets['password'])
    except ConnectionError as e:
        print("Connection Error:", e)

while True:
    print(f"Connecting... ", end="")
    wifi.radio.enabled = True
    connect()
    print(f"{wifi.radio.ipv4_address} ", end="")
    time.sleep(SLEEP)

    print(f"Disconnecting... ")
    wifi.radio.enabled = False
    time.sleep(SLEEP)
```
According to this it should print the IP as chadlung shows. That is not happening for me.

dhalbert, 2022-11-19:
I tested this with a post-8.0.0-beta.4 build with QT Py ESP32-S2 rev B and rev C boards. I am connecting to a Ubiquiti AP that is about 6 feet away. No problems. Is anyone still seeing this problem?

DJDevon3, 2022-11-19:
The Qt Py S2 I was testing with is now a permanent part of my dragon skull project. Ordered another one which should actually be arriving later today.  Will look into testing this for you tonight.  @dhalbert  should I use a nightly build or would you prefer I use the beta 4 release?

DJDevon3, 2022-11-20:
Adafruit CircuitPython 8.0.0-beta.4-38-g96fc85cd1 on 2022-11-19; Adafruit QT Py ESP32S2 with ESP32S2

```py
Connecting... 192.867.5.309 Disconnecting... 
Connecting... 192.867.5.309 Disconnecting... 
Connecting... 192.867.5.309 Disconnecting... 
Connecting... 192.867.5.309 Disconnecting... 
Connecting... 192.867.5.309 Disconnecting... 
Connecting... 192.867.5.309 Disconnecting... 
Connecting... 192.867.5.309 Disconnecting... 
Connecting... 192.867.5.309 Disconnecting... 
Connecting... 192.867.5.309 Disconnecting... 
```
Tested and working on both the Feather ESP32-S2 (no bme) and QT Py ESP32-S2 on current nightly. At least as far as this test script is concerned it's good to go.

DJDevon3, 2022-11-20:
Good job devs. Bug squashed. Closing issue.

closed 2022-11-20 by DJDevon3