adafruit/circuitpython #7128: wifi module for the Arduino Nano RP2040

CaptainHaddock2016, 2022-10-26:
The Arduino Nano RP2040 has both Bluetooth and Wi-Fi capabilities but only Bluetooth is supported my CircuitPython on this board. Are there any plans in the future to add Wi-Fi support to the Arduino Nano RP2040?

dhalbert, 2022-10-26:
The CircuitPython build for Arduino Nano RP2040 Connecthas `_bleio` turned on for coprocessors loaded with Adafruit NINA-FW (AirLift) style firmware. I don't know that it actually works with the firmware on the Nano RP2040 UBlox module. Have you successfully used it?

Similarly, NINA-FW-style CircuitPython libraries like https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI could theoretically work with the firmware on the uBlox module. But I don't know if anyone has tried it.

Since we don't make the board, we would not provide official support for it. If someone wants to work on, that would be fine.

We forked Arduino NINA-FW some time ago and have made various changes, including control-pin changes, so

anecdata, 2022-10-26:
I thought it worked, there are a number of issues and PRs from some time ago to add the board and tweak the config, and this Learn Guide:
https://learn.adafruit.com/circuitpython-on-the-arduino-nano-rp2040-connect
I think some of the pins are different, so the Arduino NINA firmware must be used, not the Adafruit fork. I seem to recall that the Arduino NINA firmware needed to be updated for some older boards. Have you tried it with the Arduino NINA firmware and the Adafruit ESP32SPI library?

HelenFoster, 2022-10-28:
The Arduino uses the Arduino NINA firmware, while AirLift modules use the Adafruit NINA firmware, but the module on the Arduino works perfectly if treated as an external AirLift module. The example in the learn guide works for me on `8.0.0-beta.3`. I previously had to update the Arduino NINA firmware to get SSL working.

CaptainHaddock2016, 2022-10-29:
> I thought it worked, there are a number of issues and PRs from some time ago to add the board and tweak the config, and this Learn Guide: [...]

closed 2022-10-29 by CaptainHaddock2016