adafruit/circuitpython #7134: RPi Pico W, Saving file returns: [Errno 30] Read-only filesystem

Keyjetter, 2022-10-26:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.3 on 2022-10-22; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
text = open("TextFile.txt", "w")
text.write("Test")
text.close()
```

### Behavior

Returns the following error:
  Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  OSError: [Errno 30] Read-only filesystem

anecdata, 2022-10-26:
The onboard CIRCUITPY filesystem is read-only by default so that code doesn't collide with edits from USB. More here:
https://learn.adafruit.com/circuitpython-essentials/circuitpython-storage
Feel free to get help on Discord https://adafru.it/discord in the #help-with-circuitpython channel

closed 2022-10-26 by jepler

Keyjetter, 2022-10-27:
Thanks anecdata. Much appreciated!