adafruit/circuitpython #7158: ESP32C3 XIAO CircuitPython 8.0.0-beta.4

WULFFJ, 2022-10-30:
### CircuitPython version

```python
ESP32C3 XIAO CircuitPython 8.0.0-beta.4
```

### Code/REPL

```python
N/A Can't upload or doing anything else, so its pointless.
```

### Behavior

Issues:

Timeout error still happening.

Thonny file upload of anything larger than read me locks up from 7-30%.

MUEditor only brings up REPL.  Files cannot be uploaded.  You have to have it on CircuitPython setting.

ESPWebTool - the newer one will get to the upload portion and freeze.

ESPWebTool (OLD) - You can upload, but its not going to work with anything.

ESPTool gets the most functionality of any solutions to do any of the above.

Can't have anything USB plugged in or any software running besides the one you are using at the time.

### Description

I have also tried just building the main branch in WSL.  The directions leave a bit to the imagination about the rest of the process.  Am I supposed to upload just one of the firmware files after it makes?  That did not work.

I really need to get this going, if you gave questions, I will be glad to assist.

dhalbert, 2022-11-02:
Did this used to work in beta.1 or beta.2?

WULFFJ, 2022-11-04:
> Did Thonny used to be able to upload larger files in beta.1 or beta.2?

No.  My whole experience with all of this has been terrible.  Even when it works, it does nothing reliable.  I have had the other Xiao boards and no issues.  Its not just Thonny either.  

It appears to be something related to pyserial from the error. \serialwin32.py", line 325, in write
    raise SerialTimeoutException('Write timeout')

I suspect something with this is causing issues with everything but the flashing of the board.  It seems to flash fine on ESPtool or the old online ESP web tool.

Anything uploading causes me issues (Thonny)
Cant ever see the files to upload in MU

Is there any other software for file management or something I can try?  I have beat my head for two weeks now with this thing.

> Did Thonny used to be able to upload larger files in beta.1 or beta.2?

No.  My whole experience with all of this has been terrible.  Even when it works, it does nothing reliable.  I have had the other Xiao boards and no issues.  Its not just Thonny either.  

It appears to be something related to pyserial from the error. \serialwin32.py", line 325, in write
    raise SerialTimeoutException('Write timeout')

I suspect something with this is causing issues with everything but the flashing of the board.  It seems to flash fine on ESPtool or the old online ESP web tool.

Anything uploading causes me issues (Thonny)
Cant ever see the files to upload in MU

Is there any other software for file management or something I can try?  I have beat my head for two weeks now with this thing.

WULFFJ, 2022-11-07:
I have found one of the big issues I was having with the XIAO esp32c3.  You have to have the antenna attached for it to work with the webserial.  It also seems to have an affect on every other means of interacting with it.  I am not sure why this is, but was able to get MU Editor to work pretty consistently.  The adafruit webserial works now and is pretty reliable.  Thonny is still not working with this.

closed 2022-11-07 by WULFFJ

dhalbert, 2022-11-07:
> I have found one of the big issues I was having with the XIAO esp32c3. You have to have the antenna attached for it to work with the webserial.

!!

camp-easy, 2023-11-15:
why my  xiao esp32c3 board doesn't get recognized by mu editor but it's ok with Thonny?

dhalbert, 2023-11-16:
> why my xiao esp32c3 board doesn't get recognized by mu editor but it's ok with Thonny?

The ESP32-C3 chip does not have full native USB. It cannot present as the CIRCUITPY USB drive. Mu only works when a USB drive is available. Thonny reads and writes files by sending invisible (to you) commands to the REPL over USB serial.