adafruit/circuitpython #7167: Seeed-XIAO-BLE-nRF52840 NON-SENSE vs SENSE

dglaude, 2022-11-04:
Seed studio has two XIAO (QT-Py format) nRF52840 board.

One is the Seeed_XIAO_nRF52840_Sense (see https://circuitpython.org/board/Seeed_XIAO_nRF52840_Sense/) that is supported in CP. The other one is/seems identical but without the microphone and without the 6 DoF IMU:
* SENSE: https://www.seeedstudio.com/Seeed-XIAO-BLE-Sense-nRF52840-p-5253.html
* NON-SENSE: https://www.seeedstudio.com/Seeed-XIAO-BLE-nRF52840-p-5201.html

I have both (don't ask) and they are delivered from factory with the same bootloader:
```
UF2 Bootloader 0.6.2-12-g459adc9-dirty lib/nrfx (v2.0.0) lib/tinyusb (0.10.1-293-gaf8e5a90) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: Seeed XIAO nRF52840
Board-ID: Seeed_XIAO_nRF52840_Sense
Date: Nov 30 2021
SoftDevice: S140 7.3.0
```

Both board work with the SENSE version of CircuitPython, and when I import board I see those sensor specific pin: IMU_INT1, IMU_PWR, IMU_SCL, IMU_SDA, MIC_PWR, PDM_CLK and PDM_DATA.

I think it is easy at runtime to figure out if you are on the SENSE or NON-SENSE board, like by scanning the I2C bus for the IMU.

I should be capable of making a separate board that just remove those meaningless pin.
I have not checked yet if they have different unique ID reserved to be able to distinguish them and avoid pre-commit error for duplicate.

The question (maybe for in the weed section of the meeting) is to know if there should be two separate board_ID and image generated for two board that are almost identical. Or can they keep the same image, but have the situation explained in circuitpython.org so that it is easier to find and not fear to install the SENSE firmware on the NON-SENSE board.

I think the situation is potentially similar to teensy41 that exist with or without the Ethernet chip (due to ship shortage). But in that case there is only one image for both... that new "without Ethernet" version is rather new... and maybe you know other board that have optional bits and pieces, and that have different firmware or share a firmware.

furbrain, 2022-11-05:
I've just written a library for the community bundle for the XIAO nrf52840 (PR pending: https://github.com/adafruit/CircuitPython_Community_Bundle/pull/121), it would be reasonably simple for me to add a function to this like
```python
def is_Sense() -> bool:
    try:
        with imu as IMU():
            return True
    except:
        return False
```

dglaude, 2022-11-05:
> I've just written a library for the community bundle for the XIAO nrf52840 (PR pending: [adafruit/CircuitPython_Community_Bundle#121](https://github.com/adafruit/CircuitPython_Community_Bundle/pull/121))

Great news, I feel it is a sensible thing to do, I guess I should try that.

My question was maybe more about the trade-off between, on one side:
* sharing one firmware for multiple hardware
* reducing the CICD computation
* reducing S3 storage
and on the other side:
* make it easier to find the firmware for your board (a CircuitPython.org/download issue)
* being able to clear support for one additional board
* avoid symbole that do not match anything on the board that could confuse the matter

Let's see how the community feel about this topic.

dhalbert, 2022-11-08:
I don't think this difference is worth making a separate board. Comparable situations is the PyBadge vs the PyBadge LC, which has fewer components, or the Adafruit Feather ESP32-S2 which optionally has a BME280 on board.

closed 2022-11-08 by dhalbert