adafruit/circuitpython #7170: Enable BLE (_bleio) for the original suffix-less ESP32

Neradoc, 2022-11-05:
An issue for reference, as I haven't found one.
This is not the same as #5926 as it's for support of the ESP32, not about the features of BLE on espressif.

dhalbert, 2022-11-08:
So you mean just turning on the current partial support?

Neradoc, 2022-11-08:
Yes, whatever is needed to make turning `CIRCUITPY_BLEIO` to 1 in `mpconfigport.mk` to work.

jepler, 2022-11-16:
It looks like simply setting it to =1 doesn't work...
```
-- Build files have been written to: /home/jepler/src/circuitpython/ports/espressif/build-adafruit_feather_esp32_v2/esp-idf
ninja: Entering directory `build-adafruit_feather_esp32_v2/esp-idf'
ninja: error: unknown target 'esp-idf/bt/libbt.a'
```

vladak, 2024-05-01:
#8776 also seeks BLE support for plain ESP32.

commit 689c6fac (2024-05-17): Add ESP BLE GATT server support

Enable BLE where we can. Switch 4MB, non-USB board default partitioning
over to a single 2MB firmware bank. Old boards override this setting to
keep the same behavior.

This also adds alpha support for the ESP32-C2 (aka ESP8684).

Fixes #5926 and fixes #7170
closed 2024-05-17 by dhalbert