adafruit/circuitpython #7171: Enabling BLE workflow on Espressif ?

Neradoc, 2022-11-05:
Do we want to implement BLE workflow on Espressif ?
It could be an alternative to web workflow or enabled by default, solving the problem of the creation of `.env` which otherwise requires serial access. And maybe it would be turned off if web workflow is enabled.

dhalbert, 2022-11-08:
Assigning to 8.0.0 to make a decision one way or the other.

ThomasAtBBTF, 2022-11-08:
For me, this is an "academic" discussion!
As long as BLE on ESP32(s3) is not working and developed as on the Adafruit ItsyBitsy nRF52840 (which is "out of stock"), I think you should work on getting _bleio and the other libs working as well as on the Nordic chip !

dhalbert, 2022-11-08:
On second thought, I don't think we can do BLE workflow given [#7170](https://github.com/adafruit/circuitpython/issues/5926#issuecomment-1074414184) 

> This is about half done. The two remaining large pieces are adding GATT server support and bonding support. Unfortunately GATT server support probably needs NimBLE modifications to allow for adding servers, characteristics and descriptors after things are running. The SoftDevice can do this so we've assumed it in our APIs ability to just `Service()` at any point.

dhalbert, 2022-11-08:
> As long as BLE on ESP32(s3) is not working and developed as on the Adafruit ItsyBitsy nRF52840 (which is "out of stock"), I think you should work on getting _bleio and the other libs working as well as on the Nordic chip !

It's in stock at the moment: https://www.adafruit.com/product/4481

ThomasAtBBTF, 2022-11-08:
[image]
Not everywhere !

dhalbert, 2022-11-16:
This would be a lot of work - deferring past 8.0.0.

ThomasAtBBTF, 2022-11-17:
Sad!

tannewt, 2022-11-17:
It is something we want to do. Just not for 8.0.

KarolDuzinkiewiczDAC, 2023-05-31:
Any ETA on this one in CircuitPython?

tannewt, 2023-05-31:
Nope, it is still blocked on #5926. Why do you want it on espressif instead of using nrf?

KarolDuzinkiewiczDAC, 2023-05-31:
Because I just bought Adafrtuit ESP32-S3 Feather :) But no worries, thanks
for the update.

jloic, 2023-09-04:
hello ! I've been working on keyboard with kmk/circuitpython for an esp32s3 and now the only thing I'm missing for going full wireless is the ble workflow. KMKfw is awesome and I'm sure this would be a terrific feature to support espressif chips for building nice bt keybs :)
Any idea if that feature is still being considered for future releases ?

Anutrix, 2023-10-29:
I wonder if https://github.com/asterics/esp32_mouse_keyboard provides some ideas or inspiration

closed 2025-03-04 by Neradoc