adafruit/circuitpython #7172: can i start_ap() without connecting to a network?

vithp, 2022-11-05:
I am using the Pico W and i want to make a portable web server. For instance after you connect to the access point you can view local website. However that i have seen, it seems impossible to just create an ap without connecting to a network first.

anecdata, 2022-11-05:
There is no AP on Pico W currently, follow any progress here:
https://github.com/adafruit/circuitpython/pull/7101

But you can run an HTTP server (or other type of TCP / UDP server) as a wifi client.

The best place for support questions like this is https://adafru.it/discord

closed 2022-11-05 by anecdata

vithp, 2022-11-05:
will it be added for the Pico W? i had seen on the documentation that the Pico W was a supported board for the wifi module>

anecdata, 2022-11-05:
Station (wifi client of an external AP) for Pico W is supported, same basic API as for `espressif` boards.

Follow the PR mentioned above for progress on AP for Pico W.

vithp, 2022-11-05:
ahh thank you i might use stations until it is available