adafruit/circuitpython #7176: adafruit_hid.mouse  on iphone - reset initial position

StillerJim, 2022-11-07:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.3 on 2022-08-29; Raspberry Pi Pico with rp2040
Board ID:raspberry_pi_pico
```

### Code/REPL

```python
import usb_hid
import board
import digitalio
from adafruit_hid.mouse import Mouse

mouse = Mouse(usb_hid.devices)
time.sleep(5) ## 5 sec delay
mouse.move(y=20)
time.sleep(5)
move.mouse(y=20)
```

### Behavior

whenever the Pico board is attached to the iphone using the usb adapter, the initial location of the pointer is the last location of the pointer.  So moving down 20, then again down 20 works as in intended. If you disconnect the pico and reconnect, it will run the same sequence moving down 20 and 20 but from the last pointer position before it was disconnected. 
I was looking for a way to reset the mouse to the center of the screen when the Pico board is first attached to the iphone, like it does when connected to an android device.

jepler, 2022-11-07:
The HID mouse that CircuitPython uses by default can only send relative motions. It's up to each device where the mouse pointer is placed when the device is first plugged in.

The first workaround that comes to mind is that you could begin by sending a series of large moves so that no matter where the mouse pointer started, it ends up in a known corner of the display, and moving from there by known increments. This should be possible just with changes to how Mouse is used.

A more advanced approach would involve defining an alternate kind of USB HID device, one which reports itself as an absolute pointing device like a touchscreen or drawing tablet. I do not have an example of this specific thing (though I've asked around) so you would have to do some discovery & learning on your own about it.

dglaude, 2022-11-07:
And I used that in this piece of code from the gist here:
https://github.com/dglaude/CircuitPython_udraw_Absolute_Mouse

The difficulty is that not all devices do accept and understand "absolute mouse" as that was the case for my Nintendo Switch. So to deal with that I just did what @jepler describe by translating into CircuitPython an Arduino library: https://github.com/dglaude/AGTD_CircuitPython/blob/main/mouseto.py
So the trick is to move the mouse in a corner, on the 0,0 side, and then remembering at all time where the mouse is located (that assume there are no other mouse and you don't touch your screen).

Please report on what works or not with an iPhone.

dhalbert, 2023-01-09:
Closing this after two months. Feel free to comment again if you want to reopen.

closed 2023-01-09 by dhalbert