adafruit/circuitpython #7177: Option to remove or custom text in Status bar & logo, change terminal font

bobricius, 2022-11-08:
Hi, I love circuit python, you are doing a great work
working on display independent terminal aps which can use "print" command to show
data in terminal style on LCD display
Unfortunately on 0.96" 80x160 display I kill half of display with blinka logo
I found command
`supervisor.status_bar.console = False`
it disable text but logo remaining and half of display is death
I read somewhere that is compilation option but I prefer mainstream distribution for Raspberry PICO instead custom build
I would also like to be able to change the terminal font without custom build.
Status bar is cool feature, would be great if I could change the text
with etc somethig like 
`supervisor.status.text = "Sending data ..."`
Thanks very much for this great project!

HelenFoster, 2022-11-09:
I agree that it's useful to be able to use the whole screen for terminal on smaller screen sizes.

Neradoc posted this in the Discord, but it's really not obvious.
https://discord.com/channels/327254708534116352/537365702651150357/1035320266852667422
```
display.root_group[0].hidden = False
display.root_group[1].hidden = True # logo
display.root_group[2].hidden = True # status bar
supervisor.reset_terminal(display.width, display.height * 2)
display.root_group[0].y = 0
```

deshipu, 2022-11-10:
You can set `display.root_group.scale=2` to make it use double pixels, which technically makes the font bigger, though it doesn't look that great. Of course you will need to adjust the size of the terminal too. Unfortunately the font is compiled into the firmware, so to use a different font, you would need a custom build.

bobricius, 2022-11-10:
Thanks Radomir, I can now read without glasses. Where do you go for that secret information? I didn't find it documented anywhere. Which font format is recommended for custom build?
Is somewhere secret setting to make display faster? (using Pico & st7789 320x240) maybe SPI speed?

deshipu, 2022-11-10:
The scale parameter of a group is documented here: https://docs.circuitpython.org/en/latest/shared-bindings/displayio/#displayio.Group.scale
The baudrate can be set when creating the FourWire bus: https://docs.circuitpython.org/en/latest/shared-bindings/displayio/#displayio.FourWire
The font can be set in the board definition like this: https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/boards/ugame10/mpconfigboard.mk#L37 -- it uses .bdf format. You can create/edit it using FontForge.

bobricius, 2022-11-11:
Custom font compilation works with brutalist-6.bdf
My font Pavlova16s.bdf give this errors:
`make BOARD=armachat
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
mkdir -p build-armachat/genhdr
GEN build-armachat/genhdr/moduledefs.h
QSTR updated
Traceback (most recent call last):
  File "/home/bobor/Desktop/circuitpython/ports/raspberrypi/../../tools/gen_display_resources.py", line 94, in <module>
    b[overall_bit // 8] |= 1 << (7 - (overall_bit % 8))
IndexError: bytearray index out of range
make: *** [../../supervisor/supervisor.mk:234: build-armachat/autogen_display_resources-en_US.c] Error 1
make: *** Deleting file 'build-armachat/autogen_display_resources-en_US.c'`
[pavlova16s.bdf.txt](https://github.com/adafruit/circuitpython/files/9993679/pavlova16s.bdf.txt)

deshipu, 2022-11-12:
Perhaps some characters in your font are different widths?

deshipu, 2022-11-12:
Sounds like a bug.

closed 2023-04-14 by tannewt