adafruit/circuitpython #7178: Download Link for CircuitPython 7.3.3 please

MZelenyElatec, 2022-11-08:
### CircuitPython version

```python
CircuitPython 7.3.3
```

### Code/REPL

```python
X
```

### Behavior

X

### Description

Download Link for CircuitPython 7.3.3 please

MZelenyElatec, 2022-11-08:
Hi jepler thank you for your quick reply. Im using Feather S3 but can't find 7.3.3 only from 8.xx on. In the past I was also using S3 and some 7.x version from the download page but now it seams that its disappeared?

jepler, 2022-11-08:
"Feather S3" is a bit ambiguous.

Unexpected Maker FeatherS3 download page has links to 7.3.3: https://circuitpython.org/board/unexpectedmaker_feathers3/

Adafruit Feather ESP32-S3 No PSRAM has links to 7.3.3: https://circuitpython.org/board/adafruit_feather_esp32s3_nopsram/

Adafruit's other Feather boards with S3 do not have links to 7.3.3, because they were added after 7.3.3: 
[Feather ESP32-S3 4MB Flash 2MB PSRAM](https://circuitpython.org/board/adafruit_feather_esp32s3_4mbflash_2mbpsram/)
[Feather ESP32-S3 TFT PSRAM](https://circuitpython.org/board/adafruit_feather_esp32s3_tft/)

closed 2022-11-12 by MicroDev1