adafruit/circuitpython #7180: BLE Advertisement Failure on Seeed XIAO ESP32-C3

a-a-crabtree, 2022-11-08:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.4 on 2022-10-30; Seeed Studio XIAO ESP32C3 with ESP32-C3FN4
```

### Code/REPL

```python
from adafruit_ble_radio import Radio
radio = Radio(channel=1)
radio.send('test')
```

### Behavior

```python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_ble_radio.py", line 134, in send
  File "adafruit_ble_radio.py", line 151, in send_bytes
  File "adafruit_ble/__init__.py", line 215, in start_advertising
_bleio.BluetoothError: Unknown system firmware error: 30
```

### Description

- Error when trying to start BLE advertisement on Seeed XIAO ESP32-C3 / Circuitpython 8.x.x
- Easiest to see when using BLE radio, but have seen same behavior any time when starting BLE advertising

### Additional information

 - Usually only seen between a hard reset and a soft reset. I.e., a soft reset will usually fix it, and it doesn't usually pop up until the board loses power or is hard reset, but it doesn't happen after every hard reset and I've seen it need multiple soft resets before it clears.
 - Have tried on multiple boards and have seen the issue on 8.0.0.alpha.1 through 8.0.0.beta.4.

dhalbert, 2022-11-15:
Do you have other Espressif boards you can test with? The BLE implementation on Espressif is limited due to some existing limitatations of the underlying `nimble` library, so we stopped working on Espressif BLE for a while. There may be remaining bugs.

a-a-crabtree, 2022-11-16:
Just tried it with a couple QT Py ESP32-C3 boards, not seeing the issue on either, so seems to be specific to the XIAO.

dhalbert, 2022-11-16:
MIght be an `sdkconfig` issue; I'll take a look later.

dhalbert, 2022-12-08:
The `sdkconfig` files between the QT Py and the XIAO have no interesting setting differences: they are tiny and just specify the LWIP name.

The other file differences also don't seem to be significant. I am at a loss to explain the issue. It could be some strapping pin issue. Do you have a pointer to the schematic for this XIAO? I cannot find it.

dhalbert, 2022-12-08:
I am going to move this forward to 8.x.x so that it doesn't block the 8.0.0 release.

RetiredWizard, 2023-01-30:
Came across this while trying to get CP running on the board.

https://files.seeedstudio.com/wiki/XIAO_WiFi/Resources/Seeeduino-XIAO-ESP32C3-SCH.pdf

RetiredWizard, 2023-01-30:
I just tried this on  8.0.0-rc.1 and didn't see the issue.  

I tried to flash the beta 3 version to confirm the error in the older version but couldn't get it to flash successfully. I'll give it a go again after I get some sleep.

RetiredWizard, 2023-01-30:
I looked at the original issue again and realized the issue didn't happen all the time so I created a code.py file with the test script and I did get the `Unknown system firmware error: 30` message once. I've been trying for a few minutes to recreate it using various power/cycle ctrl-d combinations and haven't been able to make it happen a second time.

RetiredWizard, 2023-01-30:
Okay to produce this reliably:

code.py:
```py
input('press enter to run code.py....')
from adafruit_ble_radio import Radio
radio = Radio(channel=1)
radio.send('test')
nprint('test done')
```
test.py:
```py
from adafruit_ble_radio import Radio
radio = Radio(channel=1)
radio.send('test')
print('test done')
```
Ctrl-D or powercycle
press enter to get to REPL
type: import test

evildave666, 2023-01-30:
Those code snippets produce the same error on the Luatos Core-ESP32C3 board on 8.0.0-rc.1 so not Xiao-specific:

```
press enter to run code.py....
test done

Code done running.
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 8.0.0-rc.1 on 2023-01-30; Luatos Core-ESP32C3 with ESP32-C3
>>> import test
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "test.py", line 3, in <module>
  File "adafruit_ble_radio.py", line 134, in send
  File "adafruit_ble_radio.py", line 151, in send_bytes
  File "adafruit_ble/__init__.py", line 215, in start_advertising
_bleio.BluetoothError: Unknown system firmware error: 30
```

jepler, 2023-01-30:
When behavior changes based on "run directly in code.py" vs "in imported code" I immediately suspect objects that are incompatible with the "long lived allocator", which selectively moves some objects from lower addresses to higher addresses at various times, including when a module that is being `import`ed finishes being executed. For example, it could be some native object owned by the Radio object.

RetiredWizard, 2023-01-30:
In case it's of help to someone that's more familiar with the workings of the BLE core code....

I turned on verbose BLE and this is the detailed error:
`_bleio.BluetoothError: Unknown system firmware error at common-hal/_bleio/Adapter.c:586: 30`

I believe the block of code that gets the "30" error code from Adapter.c is:
```c
    uint8_t own_addr_type;
    // Anonymous addresses are still resolvable. (Following what the NRF
    // implementation does.)
    rc = ble_hs_id_infer_auto(anonymous, &own_addr_type);
    if (rc != NIMBLE_OK) {
        return rc;
    }
```
As best I can tell the 30 code corresponds to `BLE_HS_EDISABLED`

The _bleio call to start_advertising referenced in the traceback is:
```c
            self._adapter.start_advertising(
                advertisement_bytes,
                scan_response=scan_response_bytes,
                connectable=advertisement.connectable,
                interval=interval,
                timeout=0 if timeout is None else timeout,
            )
```

dhalbert, 2023-01-30:
I looked briefly for non-long-lived objects but didn't spot any, but thanks, @RetiredWizard, that will help narrow things down.

davy39, 2024-01-09:
Hello,
Is there any update on this issue ?
Does someone have a workaroud to avoid this ?
Thanks for your work !