adafruit/circuitpython #7187: CircuitPython Mouse HID Movement Units?

gabriel-wainmann, 2022-11-10:
### CircuitPython version

```python
Hi

Have no idea what the units of the mouse movement are: pixels? relative something?
Also, when going back and forth can't get to the same place, which is strange.

Help please.
```

### Code/REPL

```python
-
```

### Behavior

-

### Description

-

### Additional information

-

dhalbert, 2022-11-10:
mouse is just emulating a regular mouse. It is just requesting to "move this many units" in X and/or Y.  This is how a regular mouse works. It's up to the host computer to interpret that. And if you move larger distances, you will invoke the acceleration algorithm on the host computer, which is perhaps why you are not going back to the same place.

dhalbert, 2022-11-10:
This kind of support q is better asked in https://adafru.it/discord or https://forums.adafruit.com/viewforum.php?f=60

closed 2022-11-10 by dhalbert

gabriel-wainmann, 2022-11-10:
Thank you @dhalbert