adafruit/circuitpython #7190: Garbage pixels start to appear on top left of screen after a few days

rdagger, 2022-11-10:
Using a QT PY ESP32-S2, an SSD1322 OLED display and CircuitPython 8.  Currently I just have a simple clock displayed although the program has other features.  I'm using the built-in RTC along with NTP.  Everything works except that after a few days of just sitting untouched as a clock, random pixels start appearing along the top left of the display.  If you draw over the pixels, they go away.  The program isn't crashing or throwing errors.
[image]

tannewt, 2022-11-15:
I wonder if these are actually bit flips in the SSD1322. I believe CP should only be updating the changing region and basically leaving the left half alone.

rdagger, 2022-11-15:
It's hard to troubleshoot because it takes at least 2 days to reproduce.  I switched from a label to a bitmap_label and so far it hasn't happened but unfortunately I have to switch back to label because the bitmap_label has too much flicker upon updates.

Performing a supervisor.reload() every night at midnight prevents the problem.

I have 4 QT PY ESP32-S2's and 4 of these SSD1322 displays.  Any troubleshoot suggestions to narrow down the cause?

rdagger, 2022-11-16:
I have a few: SSD1309 (128x64 mono), SSD1351 (128×128 color), ST7355 (128x160 color) & ST7565 (128x64 mono)
None of them are grayscale like the SSD1322 or have a 256x64 resolution.  I guess the SSD1309 would be the closest.
I'll try to set up a lab this weekend with multiple  configurations.

dhalbert, 2023-01-09:
@rdagger Were you able to reproduce this on other displays?

rdagger, 2023-01-09:
No.  I've had an SSD1309 OLED 128×64 display running for a week without issue.  I did have to modify my program to fit the smaller size.

I've also been able to get reliable results from the original SSD1322 OLED 256x64 display by performing garbage collection on every loop and automatically soft rebooting the QT PY once a day.  Since implementing these changes I've yet to see the dots appear.

One thing different about my SSSD1322 display is that it requires colstart=112 on initialization.  Not sure if this is pertinent.

I have 4 of these displays deployed in the field now so I should be able to get some good long-term test results.

dhalbert, 2023-01-10:
There was a small fix in `bitmaptools` recently here: https://github.com/adafruit/circuitpython/pull/7407/files. If you are not using `bitmaptools`, then never mind.

rdagger, 2023-01-10:
@dhalbert Not using bitmaptools.  My project is posted on my website if you would want to look at the code:
https://www.rototron.info/projects/sms-display/

dhalbert, 2024-03-08:
This is over a year old. Have you found out the problem? Does it occur with the latest CircuitPython 9 builds? I'll close for now, but we can reopen with new info.

closed 2024-03-08 by dhalbert

rdagger, 2024-03-08:
I switched from a label to a bitmap_label and the problem went away.  They've been running for over a year 24/7 without any screen garbage.