adafruit/circuitpython #7203: Display IO memory leak with label text

rdagger, 2022-11-12:
### CircuitPython version

```python
CircuitPython 8.0.0-beta.4-17-g9e94d7e75 on QT PY ESP32-S2

Every time you change the text on a label it consumes around 6.4K of free memory.
```

### Code/REPL

```python
from adafruit_display_text import label
from adafruit_ssd1322 import SSD1322
from board import D9, D17, D18, SPI
import displayio
import gc
import terminalio

displayio.release_displays()

display_bus = displayio.FourWire(SPI(), command=D9, chip_select=D18, reset=D17, baudrate=10000000)
display = SSD1322(display_bus, width=256, height=64, colstart=112)

mem_label = label.Label(terminalio.FONT, text="", color=0xFFFFFF)
mem_label.y = 4

group = displayio.Group()
group.append(mem_label)
display.show(group)

while True:
    mem_label.text = f'Free Memory: {gc.mem_free()}'
```

### Behavior

The memory decreases by about 6.4K every loop.

Neradoc, 2022-11-12:
Hi, that's not a memory leak, Circuitpython uses a garbage collector.
As the program runs memory is allocated and mem_free() goes down until it can no longer allocate, which triggers the garbage collector, which you can call in your loop with `gc.collect()` and see if the mem_free stabilizes, as it should, more or less. If it doesn't, then we can look into a memory leak.

Assigning a new text to a label means discarding the previous image representation of the label and creating a new one, so it does allocate new memory and the old goes out of scope, meaning it will be collected in due time.

rdagger, 2022-11-12:
Unfortunately, the garbage collection doesn't occur until the program is almost out of memory.  With the loop above, the 2 MB of memory is depleted in under 30 seconds.  I only noticed the usage because it is part of a much larger program that was experiencing weird memory crashes.

I will try the bitmap_label tonight.  I did insert a gc.collect() when I noticed but it just seemed like bad practice.  

Are there any other Display IO text objects optimized for fields whose text changes every second like a counter, a clock or sensor data?

Neradoc, 2022-11-12:
> Unfortunately, the garbage collection doesn't occur until the program is almost out of memory.

That is how it works. Garbage collection triggers when the code tries to allocate memory and there is no space left in the "free" memory. It does not mean that the board is out of memory, it just means that it's time to clean up. Calling it manually is a form of memory optimization that can be efficient in reducing memory fragmentation typically, if called in strategic places.

> I only noticed the usage because it is part of a much larger program that was experiencing weird memory crashes.

You might want to bring that up in our support channels, describing the crashes you are experiencing. It's a better place to find people who can help look into it to diagnose the cause and suggest solutions:
https://forums.adafruit.com/
https://adafru.it/discord

closed 2022-11-12 by rdagger

rdagger, 2022-11-13:
bitmap_label use 4K as opposed to 6.4K per loop.

```
from adafruit_display_text import bitmap_label
from adafruit_ssd1322 import SSD1322
from board import D9, D17, D18, SPI
import displayio
import gc
import terminalio

displayio.release_displays()

display_bus = displayio.FourWire(SPI(), command=D9, chip_select=D18, reset=D17, baudrate=10000000)
display = SSD1322(display_bus, width=256, height=64, colstart=112)

mem_label = bitmap_label.Label(terminalio.FONT,
                               text="", color=0xFFFFFF,
                               save_text=False,
                               anchor_point=(0, 0)
                               )

mem_label.anchored_position=(0, 4)
group = displayio.Group()
group.append(mem_label)
display.show(group)

memory_last= gc.mem_free()

while True:
    free_memory = gc.mem_free()
    mem_label.text = f'Free Memory: {free_memory}, difference: {free_memory - memory_last}'
    memory_last = free_memory
```