adafruit/circuitpython #7206: RP2040 pin drive strength

HelenFoster, 2022-11-14:
We talked about it on the Discord a few days ago but I thought I should raise it here too.

* The datasheet says "The GPIOs on RP2040 have four different output drive strengths, which are nominally called 2, 4, 8 and 12mA modes", and the default is 4mA.
* CircuitPython currently sets it to 12mA in DigitalInOut and UART, and doesn't set it for PulseOut, PWMOut or PIO.
* CircuitPython policy is normally to use max drive strength except where there's a particular reason not to?
* There could potentially be issues with capacitive loads and interference?

#1270 nrf pin drive strength discussion
https://github.com/adafruit/circuitpython/blob/8f414eb4ee4db6489c0f9eca1e441200ff3e852a/ports/raspberrypi/common-hal/digitalio/DigitalInOut.c#L105
https://github.com/adafruit/circuitpython/blob/8f414eb4ee4db6489c0f9eca1e441200ff3e852a/ports/raspberrypi/common-hal/busio/UART.c#L133

HelenFoster, 2022-11-14:
Looks like this field is not reset on deinit, so e.g. behaviour of PWMOut currently depends whether there was a DigitalInOut before.

https://github.com/adafruit/circuitpython/blob/8f414eb4ee4db6489c0f9eca1e441200ff3e852a/ports/raspberrypi/common-hal/microcontroller/Pin.c#L74

```
# Pi Pico with 220R between GP0 and GND
import board, digitalio, pwmio
p = pwmio.PWMOut(board.GP0, frequency=1000, duty_cycle=0xffff)
# 2.69V
p.deinit()
p = digitalio.DigitalInOut(board.GP0)
p.switch_to_output(True)
# 2.96V
p.deinit()
p = pwmio.PWMOut(board.GP0, frequency=1000, duty_cycle=0xffff)
# 2.96V
```

Do we want max drive strength for everything? Would it be done like in DigitalInOut?

dhalbert, 2022-11-14:
> Do we want max drive strength for everything? Would it be done like in DigitalInOut?

Yes, that should be fine. Do it like https://github.com/adafruit/circuitpython/blob/8f414eb4ee4db6489c0f9eca1e441200ff3e852a/ports/raspberrypi/common-hal/busio/UART.c#L133-L136

HelenFoster, 2022-11-14:
PIO looks complicated. Actually, if it's for everything, might it be better to set it for all the relevant pins on startup and then leave it alone in the individual IO libs?

dhalbert, 2022-11-15:
> PIO looks complicated. Actually, if it's for everything, might it be better to set it for all the relevant pins on startup and then leave it alone in the individual IO libs?

That makes sense. Are you interested in doing a PR?

HelenFoster, 2022-11-16:
At a high level I was thinking it makes sense to put this in one place, but I don't really have any idea where it would go or how you decide which pins are eligible.

eightycc, 2023-09-01:
If no one objects, I'd like to pick up this issue.

HelenFoster, 2023-09-02:
@eightycc Note that the drive strength for PIO was reduced since the previous discussion, to fix a specific use. (I was worried when I saw that, but fortunately it doesn't seem to affect my project too much.) https://github.com/adafruit/circuitpython/pull/7558

So it seems what I was saying in the OP is not the case, and we really need the general drive strength API, as you already started discussing over here: https://github.com/adafruit/circuitpython/issues/1270#issuecomment-1703614207