adafruit/circuitpython #7224: Servo update slow due to PWM channel duty cycle update busy-wait

BooleanMattock, 2022-11-17:
In the code for setting PWM duty cycle in the native **PWMOut** implementation for the RP2040
(https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/common-hal/pwmio/PWMOut.c)
 I found this:

`// Wait for wrap so that we know our new cc value has been applied. Clear `
`   // the internal interrupt and then wait for it to be set. Worst case, we`
`    // wait a full cycle.`
`     pwm_hw->intr = 1 << self->slice;`
`     while ((pwm_hw->en & (1 << self->slice)) != 0 &&  `
`           (pwm_hw->intr & (1 << self->slice)) == 0 &&  `
`           !mp_hal_is_interrupted()) { `
`    }`

If your PWM frequency is 1kHz or more, waiting a whole PWM cycle is probably not a big deal. But the the **servo** class from **adafruit_motor** uses **PWMOut**, and servos run at 50Hz.   This means this code could cause a delay of up to 20mS in setting the position of a single servo, which is a big deal when running multiple servos.

Discussed with @tannewt on the adafruit/circuitpython forum and agreed it should be raised as an issue.

jepler, 2022-11-18:
This was done as part of #5067 -- if it's important for the way PulseOut uses PWM, perhaps the 'wait for duty cycle update' code should be moved accordingly.

https://github.com/adafruit/circuitpython/pull/5067/commits/e87e1d81752605e4ef79e2a89d440c5b2b93ef58

tannewt, 2022-11-23:
I don't think this needs to be in 8.0.0 because its been this way a while. I'm not sure exactly what the right fix is.

tannewt, 2023-02-16:
Fixed by #7299

closed 2023-02-16 by tannewt