adafruit/circuitpython #7238: PWMOut Error on Raspberry Pi Pico

dbisu, 2022-11-19:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.4 on 2022-10-30; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
import board
import pwmio

led = pwmio.PWMOut(board.LED, frequency=5000, duty_cycle=0)
```

### Behavior

>>> led = pwmio.PWMOut(pin, frequency=5000, duty_cycle=0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: Expected a Pin

### Additional information

Using digitalio.DigitalIO(board.LED) works correctly.

dhalbert, 2022-11-19:
The LED pin on Pico W is not a regular pin. It is accessed through the CYW43 coprocessor and its firmware, and is not available for PWM use. Real pins like the regular GPIO physical pins _can_ be used with `PWMOut`. (The LED pin on the regular Pico works as you would expect.)

@jepler I am not sure if there's an improvable error message here or not.

dbisu, 2022-11-19:
Oh, ok.  I was porting some code from the Pico to Pico W.  This explains the difference.  I'll refactor my code that handles the LED.
Thanks.

jepler, 2022-11-19:
switch validate_obj_is_pin to use mp_arg_validate_type and make mp_arg_validate_type print the required type, so that the error is
```
>>> p = pwmio.PWMOut(board.LED)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: pin must be of type Pin, not CywPin
```
?

dhalbert, 2023-01-09:
@jepler is your last comment a suggestion? We can move this away from Support.

jepler, 2023-01-10:
Yes, I think a simple change could improve the error message

PR #7437 (2023-01-11): Improve type validation error messages, especially for pins
Fixes #7238, but does a lot more also.

- Per @jepler's [suggestion](https://github.com/adafruit/circuitpython/issues/7238#issuecomment-1320979886), expands `%q must be of type %q` to `%q must be of type %q, not %q`.
- Adds a new validator for objects in a collection: `object in %q must be of type %q, not %q`. Used a lot for pin lists.
- Adds `arg_name` to a lot of pin validation routines.
- Adds `arg_name` to SPI bus validation routine.
- Adds `arg_name` to `cp_enum_value` to use when validation fails.
- Replaces some `Expected ...`  [... cut]
closed 2023-01-11 by tannewt