adafruit/circuitpython #7244: Consider switching build directory

MicroDev1, 2022-11-21:
Use `build/board` instead of `build-board`, i.e. move boards build to a sub-directory of `build` in the root of the port.

dhalbert, 2022-11-21:
This seems fine to me. But I did a quick search for `build-` in the tree, and there are number of mentions, including possibly in submodules (`adabot`), for instance, so we'll need to change places that are assuming the old locations.

The top-level build directory (for documentation) is `_build`, which I kind of like, which is easy to spot. Maybe `_build/<board>`?

jepler, 2022-11-21:
As you do this, please consider providing
 * makefile variables that identify important targets e.g., $(FIRMWARE_ELF) $(FIRMWARE_BIN) $(FIRMWARE_UF2) $(FIRMWARE_MAP)
 * phony rules that make them (e.g., `firmware-elf: $(FIRMWARE_ELF)` marked as `.PHONY: firmware-elf`

For instance right now I have my own additions that include things like
```
.PHONY: boot
boot: build-$(BOARD)/firmware.uf2
        _douf2 "$(MOUNT)" "$<" "$(SERIAL)"
```
where _douf2 is a script that does the serial trick, locates the bootloader drive, & copies the uf2 to it. I'll have to change this rule but being able to make the dependency `boot: $(FIRMWARE_UF2)` would be a simplification that would then be resilient to future changes.