adafruit/circuitpython #7246: CP 8.0.0 beta4 missing supervisor.disable_autoreload

andreasbrett, 2022-11-21:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.4 on 2022-10-30; S2Pico with ESP32S2-S2FN4R2
```

### Code/REPL

```python
supervisor.disable_autoreload()
```

### Behavior

```AttributeError: 'module' Objekt hat kein Attribut 'disable_autoreload'```

### Description

It seems as if supervisor in CP 8.0.0 beta4 doesn't have the disable_autoreload function anymore. Will this be removed in CP8 or is it not yet included?

**Docs latest**
https://docs.circuitpython.org/en/latest/shared-bindings/supervisor/index.html

**Docs 7.3.x**
https://docs.circuitpython.org/en/7.3.x/shared-bindings/supervisor/index.html#supervisor.disable_autoreload

anecdata, 2022-11-21:
The API has changed. Try:
`supervisor.runtime.autoreload = False`

andreasbrett, 2022-11-21:
Thanks that did the trick! The documentation is a bit vague here. It really sounds like this is a read-only class.

closed 2022-11-21 by andreasbrett