adafruit/circuitpython #7250: code.circuitpython.org: navigator.bluetooth.getDevices is not a function

wwj718, 2022-11-23:
### CircuitPython version

```python
adafruit-circuitpython-microbit_v2-en_US-7.3.3.combined.hex; microbit_v2
```

### Code/REPL

```python
code.circuitpython.org
```

### Behavior

```
Uncaught (in promise) TypeError: navigator.bluetooth.getDevices is not a function

Argh: NetworkError: Connection failed for unknown reason.
```

### Description

[image]

makermelissa, 2022-11-23:
Hi, the code for the editor is in https://github.com/circuitpython/web-editor/. The BLE code on this has never worked great despite trying many things and getting BLE fixed should happen soon. I had been pulled off the project for a couple of months to get some other things done, but have resumed this week. There's a similar issue at https://github.com/circuitpython/web-editor/issues/26. I think your error message may pinpoint the an underlying issue better though.

Since the repo is owned by the circuitpython account rather than the adafruit account, I won't be able to directly transfer this issue, but I'll make a copy of it over there.

makermelissa, 2022-11-23:
New issue is here: https://github.com/circuitpython/web-editor/issues/40. I just noticed you had the microbit v2. There was an existing issue at https://github.com/circuitpython/web-editor/issues/10 from before a major code rewrite that may be related.

closed 2022-11-23 by makermelissa