adafruit/circuitpython #7251: Charge control pin missing on Seeed XIAO nRF52840

PheebeUK, 2022-11-23:
The battery charge rate control pin appears to be not defined on Seeed XIAO nRF52840 Sense. This pin controls the rate at which any attached battery is charged. This pin sets the charge current to either 50mA or 100mA. Currently there is no way to utilise the higher charge rate.

The version of CircuitPython in use is  `Adafruit CircuitPython 8.0.0-beta.4 on 2022-10-30; Seeed XIAO nRF52840 Sense with nRF52840`

Looking at pins.c, the following pins for battery interfacing are defined:
``` 
     {MP_ROM_QSTR(MP_QSTR_READ_BATT_ENABLE),MP_ROM_PTR(&pin_P0_14)},
     {MP_ROM_QSTR(MP_QSTR_VBATT),MP_ROM_PTR(&pin_P0_31)},
     {MP_ROM_QSTR(MP_QSTR_CHARGE_STATUS),MP_ROM_PTR(&pin_P0_17)},
```

However, pin `P0.13 (HICHG)` is utilised for setting the charge mode from [Seeedstudio forum: Match-san - battery charge control circuit](https://forum.seeedstudio.com/t/xiao-ble-sense-battery-level-and-charging-status/263248/6)

jepler, 2022-11-23:
Even before this is added to the board definition, that pin should be available as `microcontroller.pin.P0_13`.

tannewt, 2022-12-05:
@PheebeUK Want to add the pin into board? We can guide you to make the change. (I don't think we have the board ourselves.)

dhalbert, 2022-12-05:
@PheebeUK That's the idea. Let's call it `MP_QSTR_CHARGE_RATE` or something like that, since we are not abbreviating `CHARGE` on the other pins.

It seems hard to find documentation on this pin. I also cannot find a good schematic for the whole board. Do you have some links besides the link you gave in the first post?

PheebeUK, 2022-12-06:
@dhalbert It took me a while to find them, documentation is frustrating for this board. Schematic is available in a ZIP at [XIAO BLE Sense Getting Started](https://wiki.seeedstudio.com/XIAO-BLE-Sense-Getting-Started/#resources).

dhalbert, 2022-12-08:
@PheebeUK Shall we await your pull request?

PheebeUK, 2022-12-08:
Thanks - I would have loved to have time to deal with this, but family issues meant you got there first!

PR #7320 (2022-12-08): Add pin for charge rate for Seeed XIAO nRF52840 Sense
Fixes #7251 by adding a pin name for the battery charge control rate pin.

Pinging @PheebeUK on this.
closed 2022-12-08 by dhalbert