adafruit/circuitpython #7252: Asyncio hangs after soft reset.

m1cha1s, 2022-11-23:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.4 on 2022-10-30; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
import board
import digitalio
import asyncio

async def blink(pin, interval):
    with digitalio.DigitalInOut(pin) as led:
        led.direction = digitalio.Direction.OUTPUT
        while True:
            led.value = not led.value
            await asyncio.sleep(interval)

async def main():
    print("Started")
    t1 = asyncio.create_task(blink(board.LED, 1))
    t2 = asyncio.create_task(blink(board.GP0, 1.01))

    await asyncio.gather(t1, t2)
    print("Done")

asyncio.run(main())
```

### Behavior

After some amount of soft resets(it's seemingly random) it just hangs and does not respond or responds very very slowly. It is fixed after a hard reset(power off than on)

### Additional information

It works fine if i force the library(asyncio) to use the python version instead of _asyncio

jepler, 2022-11-23:
I think this may be #7173. In trying to make asyncio testable on host computers during the build I introduced a bug which I tried to fix in #7138 (in beta 4) but the fix was wrong; #7173 (which is not yet in the released version) I hope fixes it fully.

If you're able to grab a "very latest" build and let us know if it's working (such as https://adafruit-circuit-python.s3.amazonaws.com/bin/raspberry_pi_pico_w/en_US/adafruit-circuitpython-raspberry_pi_pico_w-en_US-20221123-3056365.uf2) it would be helpful.

Thank you for your report and any additional testing you can do!

m1cha1s, 2022-11-24:
The version you provided works without a hitch. :)

closed 2022-11-24 by MicroDev1