adafruit/circuitpython #7254: Unrecoverable crash into safe mode when calling wifi.radio.start_ap on ESP32-S3-BOX-Lite

sean-palmer-coding, 2022-11-24:
### CircuitPython version

```python
adafruit-circuitpython-espressif_esp32s3_box_lite-en_US-8.0.0-beta; ESP32-S3-BOX-Lite
```

### Code/REPL

```python
import wifi

wifi.radio.start_ap("esp32", "password123")
```

### Behavior

Board disconnects from serial connection, when code above input into REPL, it returns:

```Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
Internal watchdog timer expired.

sean-palmer-coding, 2022-11-24:
This seems to fail when just trying to connect to an AP as well.

```wifi.radio.connect("ssid", "password")```

tannewt, 2022-11-28:
What beta are you using? Please also try the absolute latest build to see if this has been fixed already. Thanks!

dhalbert, 2022-11-28:
This is a BOX Lite board. I fixed the regular BOX, but I don't have a Box Lite to test.

tannewt, 2022-11-30:
I think @makermelissa has a box lite.

sean-palmer-coding, 2022-12-05:
Just flashed the latest circuit python build available on the S3 bucket (adafruit-circuitpython-espressif_esp32s3_box_lite-en_US-20221203-a50b0a2), still the same issue.

dhalbert, 2022-12-05:
I got a BOX Lite on Friday and will be testing this.

PR #7311 (2022-12-07): make esp32-box-lite sdkconfig same as box
Fixes #7254.

The `sdkconfig` for the ESP32 Box Lite was different than the Box. Making it the same as the Box fixes the issue.

(also made the `mpconfigport.mk` the same, to be consistent)

Tested and now does not crash.
```py
Adafruit CircuitPython 8.0.0-beta.4-68-g6e40949f6-dirty on 2022-12-06; ESP32-S3-Box-Lite with ESP32S3
>>> import wifi
>>> wifi.radio.start_ap("esp32", "12341234")
>>> 
```

@sean-palmer-coding This seems to fix your problem in #7254.
closed 2022-12-07 by dhalbert