adafruit/circuitpython #7255: Building rpi pico w fails

m1cha1s, 2022-11-24:
### CircuitPython version

```python
main branch
```

### Code/REPL

```python
~/circuitpython/ports/raspberrypi$ make -j8 BOARD=raspberry_pi_pico_w
```

### Behavior

Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
CMake Error: The source directory "/home/m1cha1s/circuitpython/ports/raspberrypi/build-raspberry_pi_pico_w/pioasm" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
Makefile:66: recipe for target 'build-raspberry_pi_pico_w/pioasm/pioasm/pioasm' failed
make: *** [build-raspberry_pi_pico_w/pioasm/pioasm/pioasm] Error 1
make: *** Waiting for unfinished jobs....

### Additional information

The raspberrypi pico compiles just fine but pico w fails with the error mentioned above.

bill88t, 2022-11-24:
Did you `make fetch-submodules` from the root of the repo (`~/circuitpython`)?

jepler, 2022-11-24:
I'd also check for an incompatible make version: https://github.com/adafruit/circuitpython/issues/7117

We haven't determined a compatibility range or minimum version for cmake, but I have `cmake version 3.18.4`.

What cmake and (g)make versions do you have?

The error message says that "The source directory" doesn't exist but the path it gives has the form of the target directory (because it contains "/build-"). The full output of `make V=2 -j1 -f Makefile BOARD=raspberry_pi_pico_w build-raspberry_pi_pico_w/pioasm/pioasm/pioasm` may be helpful, if you can include it in this issue.  Here, where that's working, it leads off with
```
cmake -S pioasm -B build-raspberry_pi_pico_w/pioasm
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
```
showing the source (-S) and build (-B) locations that I expected based on the Makefile.

rperwinski, 2022-12-03:
I had the same symptoms when building in Debian:
`make BOARD=raspberry_pi_pico_w`

`Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
make: cmake: No such file or directory
make: *** [Makefile:95: build-raspberry_pi_pico_w/pioasm/pioasm/pioasm] Error 127`

I have run:
`sudo apt install cmake`

cmake version 3.18.4

and now I'm able to build it.

dhalbert, 2023-01-09:
I added `cmake` to the Building CircuitPython guide prerequisities. Thanks @rperwinski.

closed 2023-01-09 by dhalbert