adafruit/circuitpython #7260: Magtag "no network with that SSID" - 9 cases out of 10 on one device. Another OK!

tarmor1, 2022-11-26:
### CircuitPython version

```python
adafruit-circuitpython-adafruit_magtag_2.9_grayscale-en_GB-7.3.3 ,
latest magtag ,
Adafruit CircuitPython 7.3.3
```

### Code/REPL

```python
import ipaddress
import ssl
import wifi
import socketpool
import adafruit_requests
import time
from adafruit_magtag.magtag import MagTag
import adafruit_minimqtt.adafruit_minimqtt as MQTT
from secrets import secrets
magtag = MagTag(

)

[...bunch of nonrelevant constants ...]

if use_live:
  try:
    wifi.radio.connect(secrets["ssid"], secrets["password"])
    pool = socketpool.SocketPool(wifi.radio)
    requests = adafruit_requests.Session(pool, ssl.create_default_context())

[... bunch of other stuff ...]
```

### Behavior

 "no network with that SSID"

### Description

Hi all!
Opening another instance of "no network with that SSID", looks similar to https://github.com/adafruit/circuitpython/issues/4093

I have 2 identical Magtags, both brand new running the identical codebase (with just one byte difference for mqtt publish difference), same libs, same bootloader, same location in house.

One of the magtags, in 9 cases out of 10, is throwing the SSID error. Another - no errors _at all_.

The code itself essentially reads 5-6 REST API endpoints, current time from your API and sends it's own battery status to MQTT. Yet it's not too relevant as error is thrown just on `wifi.radio.connect`

What I can state about the fault: 
* with or without USB cable - the same
* further/closer of AP - the same. In fact, I'm looking at the RSSI signal that I print out with `wifi.radio.ap_info.rssi`, it's as well quite identical to the working Magtag. Only difference is another AP channel. Yet again, the issues were already there once the Magtag was closer to the other AP. (I'm running Unifi if anybody cares).
* it's coming in quite quickly, ca 5-6 sec after reboot (if passing 10 sec it's safe to say it'll boot and not catch in exception.
* I don't see any difference in hard restart vs code restart `magtag.exit_and_deep_sleep(15) #retry` vs button turn off-on.

I'm thinking it might be a hardware issue. Any pointers what in addition I could check and debug?
(Eg I haven't tried isolating APs and disabling other APs, nor CircuitPython 8).

Just for artistic visualisation of the situation:
[image]

dhalbert, 2022-11-26:
1. Try 8.0.0-latest beta or even "Absolute Newest".
2. Is it the case that the "bad" one seems bad when the "good" one is powered off?
3. Try erasing the entire flash on the bad one with `esptool.py erase_flash`, and then reload the UF2 bootloader: https://learn.adafruit.com/adafruit-magtag/install-uf2-bootloader

tarmor1, 2022-11-29:
Thanks! Opened topic https://forums.adafruit.com/viewtopic.php?t=196529