adafruit/circuitpython #7270: Add functionality to use double quotes (") in .env files

kattni, 2022-11-28:
Currently, all strings assigned to variables in the .env file must be wrapped in single quotes (`'`). I feel we should add the ability to use double quotes (`"`) as well.

I am consistently creating .env files using double quote (`"`) wrapped strings, because that is what I'm used to with CircuitPython/Python. There are a couple of problems here. One: the code fails with really obtuse errors when the .env file is not formatted properly, which leads to lengthy unnecessary troubleshooting, and basically hoping someone involved remembers to check the .env file. Two: I have an apostrophe in my WiFi SSID, and when formatted properly, I have to remember to escape the apostrophe or it obviously fails to connect. This is simply an annoyance, but I'm not sure how much escaping characters is common knowledge, and I could see others not understanding how to fix this issue.

Essentially, the big picture is that troubleshooting errors in the .env file formatting is not straightforward and could lead to support issues as it becomes more common. It fails clearly when the strings are flat out wrong (e.g. SSID being wrong, or AIO key being incorrect). But, it fails vaguely when the issue is formatting. This is, perhaps, not the entire solution to that issue, however, it does eliminate one very obvious formatting issue.

I also realise that documentation will help, but I'm uncertain whether we've pointed out that it must be single quotes in any of our documentation. If it's there, I might have missed it. The Learn documentation is still sparse, as the .env file is a relatively recent development.

I've personally been using a keys.py file, with essentially the same format as .env, except with the ability to use double quotes. I import the variables, and use them in my code. I'm not suggesting we stop using the .env file, but I'm pointing this out as it's my workaround for not dealing with the issues I get using the .env file.

tannewt, 2022-11-29:
FWIW, I think I skipped double quotes since it has more escape sequences ([format spec](https://github.com/theskumar/python-dotenv#file-format)).

I agree the failure mode isn't clear. I'm not sure where to show an error though because the file is usually parsed outside of user code.

dhalbert, 2022-12-01:
Closing: this is now subsumed by #7274.

closed 2022-12-01 by dhalbert