adafruit/circuitpython #7271: Change back the number of available sockets on S2 ?

Neradoc, 2022-11-28:
`CONFIG_LWIP_MAX_SOCKETS` was changed (accidentally) to 8 for all Espressif builds in the default sdkconfig  #6181 
Can it be the cause of potential memory issues, especially on smaller RAM builds ?
Should it be changed back (to 4) on S2 ? Or C3 ?

anecdata, 2022-11-28:
Some background [here](https://github.com/adafruit/circuitpython/issues/5915) (and in the PR). It's slightly more involved than just processor variant... whether there's PSRAM, for example to take the heap. Plus, we now have the capability for reserved espidf memory in PSRAM with `CIRCUITPY_RESERVED_PSRAM=` in `.env`.

Neradoc, 2022-11-28:
I completely missed `CIRCUITPY_RESERVED_PSRAM`, do we want to make the max sockets more flexible in that case ? With a higher hard limit and a separate soft limit checked in code that can be changed with an environment variable ?

dhalbert, 2022-12-05:
@Neradoc have you heard of memory issues on smaller boards?

Neradoc, 2022-12-05:
No, I have seen nothing that seems directly linked to this. I wanted to look around at issues raised since the change but I got nothing. I think we can close this and revisit it if and when it becomes an issue.

anecdata, 2022-12-05:
Agreed. It's an edge case: lower-SRAM devices (ESP32-S2 and maybe ESP32-C3) boards with no PSRAM and trying to use an unusual amount of espidf memory.

tannewt, 2022-12-06:
Thanks for the discussion! We can open another issue if we change our minds later.

closed 2022-12-06 by tannewt