adafruit/circuitpython #7274: Consider using something other than dotenv in 8.0.0

dhalbert, 2022-11-29:
(Motivated in part by #7270 and in part by my own experiences.)

I have been thinking about alternatives to dotenv. I have seen these issues with the current scheme:

- The dotenv syntax is poorly specified and isn't consistent between Javascript, Python, shell, and other implementations.
- To the beginning user, the syntax is unique, different, and idiosyncratic.
- We don't implement the whole syntax anyway.
- The syntax is hard to parse properly, and uses up more code space than a simpler syntax. Consider the problems we've had with leading and trailing spaces, and comment indictators.
- The `.env` filename with the leading `.` is not handled well by some editors, by Windows and other OS's, and is invisible if hidden file viewing is not turned on in the host OS.

I am thinking of these possibilities:

- Use functionality in `boot.py`, such as `os.environ`, to set some persistent values. Could the web and BLE workflow startups be deferred until after `boot.py` runs?
- Implement a subset of `.ini` or `.toml` syntax`: one section only, no nesting.
- Invent our own syntax, maybe something like:
  - Just assignments, without quotes
  - A multiline value would use `\n`. It would not actually be multiline in the file.
  - `\\` escapes `\`. 
  - `# Comment` only permitted on a line by itself.
  - not a `.file` naming scheme

It is not too late to make a change like this: we are still in beta.

dhalbert, 2022-11-29:
Maybe I am just being too grumpy. But the details of the `sh`-like syntax for dotenv are not familiar to many, and surprise even experienced people.

There is `configparser` in CPython, which does something like INI files, but its API requires reading something to parse and then fetching values from the object of parsed values. That is not so convenient.

We could narrow what is allowed in `.env` files:
- all values must be in double quotes: no bare values, no single quotes. Or allow single quotes and double quotes, with restrictions on escapes. Requiring quotes solves the ambiguity of whether a `#` begins a comment or is part of a value. 
- implement only `\n`, \`r`, `\\`, and `\"` escapes for double quotes. Other escapes are an error.
- no multiline values: use `\n` instead
- keys may not be quoted

In addition, we need an alternate name for the `.env `file, such as `environ.env`, so that creating the `.env` file is easier, and it's not sometimes a hidden file.

jepler, 2022-11-29:
I don't think you're being grumpy, I think you're looking out for the interests of users. If you were being grumpy, you'd demand that it match ksh93's parsing of single-quoted strings exactly :rofl: (that's me, I'm that person)

I'd probably call the file `settings.txt` or `cpsettings.txt`. I would probably suggest NOT having a bunch of alternative filenames.

It'd be nice if we could avoid creating our own language or our own parser API.

Implementing os.setenv() is probably hard, because now you have to be able to carry that key-value dictionary in RAM and across VM resets. And since we want to bring up wifi workflow before boot.py, it doesn't help to store wifi credentials.

If we do, though, a simple key-value subset of yaml looks like it would have a similar code complexity to dotenv. If the API is only `os.getenv()` it is fine that it doesn't match the common PyYAML library API. (that is, withdraw the `dotenv` _module_ altogether)

```yaml
# comment is ignored and can contain an unmatched "
WIFI_SSID: Guest WIFI
WIFI_PASSWORD: "Speak \"friend\" and enter"
```

Are hex and/or unicode escapes actually important? I have heard of, but never tried to use, wifi SSIDs that contain non-ASCII.

One incompatibility I spotted is that "FOO: 3" would result in an int from `yaml.load` but of course would be a string in this implementation. So, an additional unenforced rule would be that a value that starts with a digit must be quoted, if yaml-compatibility is required.

pure Python implementation; the C implementation would be more like our dotenv implementation now in structure (I don't know that there's a substantial size savings):
https://gist.github.com/jepler/5cdb4235cec09232402bff70bfc79dcd

jepler, 2022-11-29:
toml doesn't have *any* unquoted string values, but it does have "=" rather than ":" syntax.

dhalbert, 2022-11-29:
Yes, typed values is why I avoided YAML. INI, restricted .env, or TOML are all strings only. Strings only has actually been a problem when people want to specify boolean values. There's no consistent mapping. General primitive-typed config values conflict with environment variables.

> toml doesn't have any unquoted string values, but it does have "=" rather than ":" syntax.

Only issue with TOML and INI are the section names, but we could say we ignore section names for now, or just say it has to be `[ENV]` or something like that.

anecdata, 2022-11-29:
Since wifi credentials are a key element of this...

Wi-Fi SSIDs can most definitely contain unicode, etc. In fact, an SSID can (in theory) be _any_ bytes, there is no restriction in spec. It is up to the receiver of that SSID to parse it as intended (or not).

jepler, 2022-11-29:
Wikipedia says that the TOML "specification includes a list of supported data types: String, Integer, Float, Boolean, Datetime, Array, and Table."

inifile is extremely ill-defined; there's no standard for quoting or escape sequences. Heck, it's not even universal whether '#' or ';' is the comment character.

HelenFoster, 2022-11-29:
I saw a comment on the Discord that it was changed from secrets.py to .env because secrets.py clashes with a CPython module called secrets. And here people are saying it can't go in boot.py because wifi workflow starts before that. But would it be an option to use a different py file in some way? Users already know the syntax, and the parser is implemented already.

jepler, 2022-11-29:
Right now, because of some design decisions in CircuitPython, it's either "really inconvenient" or "maybe actually impossible" to use regular Python code to set the Web Workflow wifi details & the Espressif "reserved PSRAM" details. That's the main driver of adding some kind of new-to-circuitpython file format in 8 for these things; it's a convenient coincidence that it can also be used via `os.getenv` and/or `dotenv`.

Thinking about this since I saw the issue filed last night, and given the discussion on Discord, I think we would be much better off choosing a format that is an easy-to-explain subset of Python is a good way to go. The filename should be one that is shown in standard file dialogs and opens easy in standard editor programs, such as "cpsettings.txt". We remove barriers for our users when they
 * do not have to change operating system settings before opening the file in a file chooser
 * do not have to learn a different way of writing quoted strings than the one they use in Python code

The values would be available to Python code via `os.getenv()` only; the `dotenv` module would be withdrawn

File format, hand-wavy style:
 * Each line may be empty, a comment, or of the form "NAME = VALUE"
 * A line is empty if it has no non-whitespace characters
 * A line is a comment if it the first non-whitespace character is "#"
 * NAME is a Python identifier and must begin at the first column
 * Whitespace is permitted around "=" and after the closing quote of VALUE
 * VALUE is always enclosed in double quotes `"..."`. Single quotes, triple quotes, and double-triple quotes are not supported. b-, f-, r- and u-string syntax is not supported
 * "\\", "\"", "\r", "\n", "\t", and "\uXXXX" escapes are supported

We might also try again to figure out what it would take to be able to code these things in regular old Python (via boot.py) instead of introducing a new file and new file format. 1 to 2 days of time spent on that might allow us to just get rid of the whole question. (even though I kinda like the declarative way of doing it, it creates a filename and file format we have to carry forever or at least for a long time)

dhalbert, 2022-11-29:
> Right now, because of some design decisions in CircuitPython, it's either "really inconvenient" or "maybe actually impossible" to use regular Python code to set the Web Workflow wifi details & the Espressif "reserved PSRAM" details.

In `main.c`, I think it might be possible to call `supervisor_workflow_reset()` (which also starts the radio-based workflows) only after `boot.py` runs. Or the reset could be called with a bool arg that determines whether to actually start the workflows or just reset the state. Right now `supervisor_workflow_start()` is also called _after_ `boot.py` runs.

Does the PSRAM reservation need to be done before `boot.py` runs, or could it be reserved afterwards, before `code.py`, as long as the value is known?

The contents of `os.environ()` could be preserved in a `storage_allocation` or via other means when the `boot.py` VM finishes. Right now the USB setup done in `boot.py` is preserved by copying the data onto the stack temporarily and then copying it back. But I might not have understood how to use a movable storage allocation for the same purpose.

> File format, hand-wavy style: ...

@jepler, I am thinking more or less the same thing, though more as a restriction on .env than as restricted Python syntax. But it comes out to the same syntax. `\uxxxx` would be handy, and it's not in dotenv that I know of, at least not officially.

jepler, 2022-11-30:
At one point the psram allocation had to happen before circuitpython heap allocation. I'm not sure if that's true now.

jepler, 2022-11-30:
That is, before any circuitpython heap was ever allocated. I think that's not true anymore, because it's done via a supervisor allocation.

dglaude, 2022-11-30:
For alternative name to `.env` I would like to suggest `dotenv‘ (or `dotenv.txt` if this is easier for Windows user using notepad) just because that is funny but easy to remember.

dglaude, 2022-11-30:
Just a few ideas:

* Whatever the name/syntax, I would love to see adafruit.io to be updated so that recovering you keys and storing them for CircuitPython is as simple as possible.
* Please also consider to also look at wippersnapper that has it's own configuration file that contain the same things: ssid+password+adafruit.io keys+timezone.
* Maybe wippersnapper should be able to detect and use the same file that CircuitPython is using...
* Also, don't forget the quantity of learn guide that explain current way of dealing with secrets.
* And if git ignore is already used somewhere (libraries, projects), make sure you add the new secret files.

tannewt, 2022-11-30:
Here are some random thoughts from me:

`.env` is the most common way for storing secrets with CPython and containers that I've found. I think matching that has value. I'd target compatibility with CPython's version only so that Blinka examples can use the same code. I'd be ok switching to double quotes only. It was my intention to only handle a subset already since we don't do the substitution stuff.

I wouldn't want to invent our own syntax because then we don't get any existing tooling benefits.

JSON could be an option since we already have parser. Wippersnapper uses it for credentials now too. The trick with any parser is to make it work without any allocations. I'm not sure the existing JSON one would. JSON also has no way to add comments. It is almost Python dictionaries in syntax too.

I like TOML and would be happy to see us have native parsing support. It supports comments and is well defined. We could have a single section we use for CircuitPython and the rest of the sections could be for the user.

I think users would love to be able to set `os.environ` and have it persist across runs.

I think it'd be ok to start all workflows after boot.py. It really was meant to control what CP does later before `code.py`.

dhalbert, 2022-11-30:
> > I would love to see adafruit.io to be updated so that recovering you keys and storing them for CircuitPython is as simple as possible.
> 
> @dglaude Is there a method/technique you've liked so far and have felt is **simple**?

This is a good discussion but should probably be held in another place.

jepler, 2022-11-30:
I would be happy to prototype and pull request a "toml subset" parser in C. It's similar to ideas discussed above.
 * the filename is "/settings.toml"
 * on any board that supports `os.getenv` via this file, `/settings.toml` is created as empty when CIRCUITPY is formatted
 * the content is assumed to be in UTF-8 encoding
 * the supported data types are string and integer
 * only basic strings are supported
 * integer support to be determined. It may be only integers supported by `strtol`. (no `0o`, no `0b`, no underscores `1_000`, `011` is 9, not 11)
 * only bare keys are supported
 * duplicate keys are not diagnosed.
 * keys with string values can be retrieved by os.getenv()
 * only values from the "root table" can be retrieved (parsing ends when it encounters a line starting with `[`)
 * "\U" (unicode non-BMP) escapes not supported. unicode values \uFFFF and below would be available. Other \-codes also supported
 * the dotenv importable module is withdrawn
 * no toml-parsing importable module is added; a fuller-featured toml parser could be implemented in pure python, or basic data can be retrieved via `os.getenv`

This would make the following `settings.toml` file valid:
```toml
CIRCUITPY_WIFI_SSID = "Guest wifi"
CIRCUITPY_WIFI_PASSWORD = "rainbowtable328"
```

Because triple-quoted strings would not be understood, the following pathological but valid toml file would have the key "oops" to the mini-toml implementation but not to a proper one:
```toml
TRICKY=""""
oops=1
```

Bonus if it can
 * warn about non-supported constructs, at least triple-quoted strings
 * finesse a way to allow that warning to appear in boot_out.txt if it occurs while retrieving a value during early startup

dhalbert, 2022-11-30:
This seems excellent to me. I didn't realize TOML had a "root table", which makes things simpler.

In the long run we can add `os.environ()` support, so that these persistent values could be computed in `boot.py`. But we don't have to do that now. This proposal nicely simplifies explanations of the format, parsing, and file creation on OS's that don't deal with dotfiles well.

tannewt, 2022-11-30:
Works for me. :+1:

jepler, 2022-12-02:
Status: in my work in progress branch, a bunch of test cases pass. However, the internal interface has changed and so the shared-modules that used dotenv directly need to be updated.

```
# comment
string = "hello world"
number = 7
 cstring = "hello comment" # comment
  cnumber = 0x7f # comment
string1= "\n"
string2 ="\u00c1x"
string3 = "\U000000c1x"
string4 = "\f\"\\"
string5 = "\t\r\b"
```
My work is pushed to a branch, but I'll be waiting on creating a PR until it's closer to completion: https://github.com/jepler/circuitpython/pull/new/dotenv-becomes-toml

PR #7321 (2022-12-14): Dotenv becomes toml
This PR replaces the `dotenv` module and the `.env` file with the `_environ` module and the `settings.toml` file.

Testing performed:
 * on pico w, set up wifi workflow with alternate port number
 * ran environ_test in build and on pico w

Would appreciate test reports of scenarios I didn't test, like
- [x] ble workflow
- [x] espressif wifi workflow

closes #7274
closed 2022-12-14 by dhalbert