adafruit/circuitpython #7279: -O2 optimization is broken on samd5x/e5x

jepler, 2022-11-29:
### CircuitPython version

```python
8.0.0-beta.4-50-g4af95f1cb1 (local build) on grand central m4
```

### Code/REPL

```python
no code.py
```

### Behavior

When the Makefile is changed to use -O2 instead of -Os, USB CDC immediately disconnects.  Safe mode doesn't help.  mass storage doesn't mount.

This is similar to #7277 except on SAMD51; it doesn't show up in default builds because they're using -Os optimization.

dhalbert, 2022-12-23:
Testing on Grand Central M4. CircuitPython crashes quickly, after only a few seconds. It looks like it's due to a CIRCUITPY read request via USB. Backtrace, showing a memory error causing a hard fault when trying to read QSPI data.
```py
#0  reset_into_safe_mode (reason=HARD_CRASH) at ../../supervisor/shared/safe_mode.c:122
#1  0x000077b4 in HardFault_Handler () at supervisor/port.c:723
#2  <signal handler called>
#3  memcpy (dst=0x20001968 <_mscd_buf.lto_priv.0>, src=src@entry=0x23fefe00, n=n@entry=512)
    at ../../shared/libc/string0.c:70
#4  0x0006dc6c in spi_flash_read_data (length=512, data=<optimized out>, address=536804864)
    at supervisor/qspi_flash.c:208
#5  read_flash.part.0.lto_priv.0 (address=536804864, data=<optimized out>, data_length=512)
    at ../../supervisor/shared/external_flash/external_flash.c:87
#6  0x0003b27e in read_flash (data_length=512, data=0x20001968 <_mscd_buf.lto_priv.0> "\003", address=536804864)
    at ../../supervisor/shared/external_flash/external_flash.c:81
#7  external_flash_read_block (block=1048447, dest=0x20001968 <_mscd_buf.lto_priv.0> "\003")
    at ../../supervisor/shared/external_flash/external_flash.c:524
#8  supervisor_flash_read_blocks (num_blocks=2, block_num=<optimized out>, dest=<optimized out>)
    at ../../supervisor/shared/external_flash/external_flash.c:576
#9  flash_read_blocks (dest=<optimized out>, block_num=<optimized out>, num_blocks=2)
    at ../../supervisor/shared/flash.c:113
#10 0x0002fbd4 in mp_vfs_blockdev_read (buf=0x20001968 <_mscd_buf.lto_priv.0> "\003", num_blocks=<optimized out>,
    block_num=<optimized out>, self=<optimized out>) at ../../extmod/vfs_blockdev.c:52
#11 disk_read (pdrv=<optimized out>, buff=0x20001968 <_mscd_buf.lto_priv.0> "\003", sector=<optimized out>,
    count=<optimized out>) at ../../extmod/vfs_fat_diskio.c:42
#12 0x0006974e in tud_msc_read10_cb (buffer=<optimized out>, offset=<optimized out>, bufsize=<optimized out>,
    lba=<optimized out>, lun=<optimized out>) at ../../supervisor/shared/usb/usb_msc_flash.c:178
#13 proc_read10_cmd.constprop.0 (rhport=<optimized out>, p_msc=<optimized out>)
    at ../../lib/tinyusb/src/class/msc/msc_device.c:843
#14 0x00059d66 in mscd_xfer_cb (rhport=<optimized out>, ep_addr=<optimized out>, event=<optimized out>,
    xferred_bytes=<optimized out>) at ../../lib/tinyusb/src/class/msc/msc_device.c:440
#15 0x00083644 in tud_task () at ../../lib/tinyusb/src/device/usbd.c:570
#16 0x00085072 in tud_task () at ../../lib/tinyusb/src/tusb.c:63
#17 usb_background () at ../../supervisor/shared/usb/usb.c:203
#18 usb_background () at ../../supervisor/shared/usb/usb.c:200
#19 usb_background_do (unused=<optimized out>) at ../../supervisor/shared/usb/usb.c:220
#20 0x00016be0 in background_callback_run_all () at ../../supervisor/shared/background_callback.c:97
#21 0x000762d6 in background_callback_run_all () at ../../supervisor/shared/background_callback.c:81
#22 supervisor_run_background_tasks_if_tick () at ../../supervisor/shared/tick.c:122
#23 run_code_py (simulate_reset=<synthetic pointer>, safe_mode=<optimized out>) at ../../main.c:536
#24 main () at ../../main.c:1019
```
In `memcpy()`, I tried turning off the multi-byte copy optimizations, which did not help. The problem appears to be further down.

if I write a blink program, and reset while connect to a host, blink runs for 2 or three blinks, and then CPy crashes. If I connect to an AC adapter, so there's no USB connection, it runs indefinitely.  So in both cases, the internal reading of flash to get the program is working fine. The problem is the USB read.

dhalbert, 2022-12-27:
I know what the basic problem is. Working on a fix. It has to do with the external flash setup not being finished before the USB host asks for the MSC size.

dhalbert, 2022-12-29:
This ultimately appears to be due to not telling gcc to do a software memory barrier after turning off the SAMD cache.  In `-O2` code, inlining some routines made this necessary. PR soon. I am seeing whether I also need to do hardware memory barrier instructions also, and what the situation on other ports is.

I found out one interesting thing: you can adjust gccoptimization levels per function inside a source file:
https://gcc.gnu.org/onlinedocs/gcc/Function-Specific-Option-Pragmas.html#Function-Specific-Option-Pragmas
```c
#pragma gcc optimize("Os")
// ... some function definitions
#pragma gcc reset_options
```
(Note that there is not `push` and `pop`, which is used for diagnostics `pragma`s).

This allowed some fine-grained testing of exactly which functions were affected negatively by the optimization level.

PR #7398 (2022-12-30): Use memory fence when disabling cache to avoid -O2 problems
Fixes #7279.

Disabling the cache on SAMx5x caused problems when compiling with -O2, apparently due to inlining the microcontroller cache control routines.. Access to the external flash chip did not work.

Changes made:
- https://github.com/adafruit/samd-peripherals/pull/43 adds a memory fence before and after disabling the cache.
- `external_flash.c` tries a little harder to make sure the flash chip is ready, by retrying fetching the JEDEC ID if necessary. Without this, if the chip was not actually ready, the ID would not be read, an [... cut]
closed 2022-12-30 by dhalbert