adafruit/circuitpython #7284: BLEConnection.disconnect() doesn't work on nRF52840

xorbit, 2022-11-30:


dhalbert, 2022-12-07:
@xorbit The BLE workflow is a way of editing files and talking to the REPL over BLE. You use the File Glider or PyLeap iOS apps to use the workflow.

I just want to narrow this down. Thanks.
1. Did you ever user either of those apps?
2. Do you have an `.env` file on your nRF board? If so, what's in it (minus private info)?
3. If you "forget" the CPB board on the host device (e.g. a phone), do you have this problem?
4. If available, could you point me to the entire code for your program?

xorbit, 2022-12-07:
Hi Dan,

No, we've never used the BLE workflow.  We've only ever updated the scripts over USB.  There is no .env file on the board and since we didn't use the BLE workflow there is nothing to forget on the phone.  The code is pretty much the plain Adafruit BLE HID example as far as BLE is concerned with the code above added to attempt a disconnect when a key is held on bootup.

tannewt, 2022-12-08:
The BLE workflow is on by default. Only wifi needs .env. You can disable ble workflow with `supervisor.runtime.ble_workflow = False` iirc.

I believe HID does require pairing and bonding so you'll be have data saved on both ends of the connection. After you disconnect, do you expect to reconnect to that device again without pairing? If not, you could call `_bleio.adapter.erase_bonding()`. This is what CP does when the reset button is pressed during the fast blink on startup.

dhalbert, 2022-12-08:
The BLE workflow looks for a paired/bonded device and connects to it on startup. It is true the HID also requires pairing/bonding, for security reasons, so it sounds like in this case the BLE workflow code can't tell the difference. So, as @tannewt mentions, doing `supervisor.runtime.ble_workflow = False` early in your code (possibly in `boot.py` will turn off the BLE workflow, and then your disconnect should work properly.

From https://docs.circuitpython.org/en/latest/shared-bindings/supervisor/index.html#supervisor.Runtime.ble_workflow:

> **ble_workflow: _bool_**
> Enable/Disable ble workflow until a reset. This prevents BLE advertising outside of the VM and the services used for it.

I hope this works for you. The host device that is doing workflow may also be a device you want to connect with using HID, so I'm not sure how we can distinguish them.

xorbit, 2022-12-09:
I am on CircuitPython 7.3.3 currently, which doesn't seem to have the `supervisor.runtime.ble_workflow` so I'll have to try that after upgrading some other time.

I tried the `_bleio.adapter.erase_bonding()` and it gets the device into a weird state where it seems to cycle between connected and disconnected state continuously.  Very odd.

dhalbert, 2023-10-25:
Closing due to inactivity, but we can reopen if this is still an issue for you.

closed 2023-10-25 by dhalbert