adafruit/circuitpython #7300: deep sleep works the first time it is called for the set time and then the microcontroller always wakes up again immediately

BeatArnet, 2022-12-03:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.4-57-ge82a8bf8b on 2022-11-30; Adafruit Feather ESP32-S3 TFT with ESP32S3
```

### Code/REPL

```python
# Powering down TFT-display and sensors
p.value = False 

# Create a an alarm that will trigger 120 seconds from now.
time_alarm = alarm.time.TimeAlarm(monotonic_time=time.monotonic() + 120)
# Exit the program, and then deep sleep until the alarm wakes us.
print ("120 seconds in deep sleep")
alarm.exit_and_deep_sleep_until_alarms(time_alarm)
# Does not return, so we never get here.
```

### Behavior

First time after booting: waits 120 seconds in deep sleep
Second time and the following: immediately wakes up again and starts

tannewt, 2022-12-05:
I'm not seeing this behavior on an ESP32-S3 USB OTG board. Here is my bootloader output showing repeated 2 minute deep sleeps:

```
[00:02:06.793] ESP-ROM:esp32s3-20210327
[00:00:00.006] Build:Mar 27 2021
[00:00:00.000] rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
[00:00:00.005] pro cpu reset by JTAG
[00:00:00.000] SPIWP:0xee
[00:00:00.000] mode:DIO, clock div:1
[00:00:00.000] load:0x3fce3808,len:0x68
[00:00:00.005] load:0x403c9700,len:0x980
[00:00:00.000] load:0x403cc700,len:0x25a0
[00:00:00.005] SHA-256 comparison failed:
[00:00:00.000] Calculated: ba9b2d4eb238d33c8b2df1f10920a53b5a565dd3a0a23ff135232249c77b4a85
[00:00:00.005] Expected: bd6e2eb493ec8e2143cb0d39c23b5c2005f98ed268d8910021afbdfc58c4fde7
[00:00:00.012] Attempting to boot anyway...
[00:00:00.000] entry 0x403c98a4
[00:02:06.732] ESP-ROM:esp32s3-20210327
[00:00:00.007] Build:Mar 27 2021
[00:00:00.000] rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
[00:00:00.005] SPIWP:0xee
[00:00:00.000] mode:DIO, clock div:1
[00:00:00.000] load:0x3fce3808,len:0x68
[00:00:00.005] load:0x403c9700,len:0x980
[00:00:00.000] load:0x403cc700,len:0x25a0
[00:00:00.000] SHA-256 comparison failed:
[00:00:00.005] Calculated: ba9b2d4eb238d33c8b2df1f10920a53b5a565dd3a0a23ff135232249c77b4a85
[00:00:00.005] Expected: bd6e2eb493ec8e2143cb0d39c23b5c2005f98ed268d8910021afbdfc58c4fde7
[00:00:00.005] Attempting to boot anyway...
[00:00:00.004] entry 0x403c98a4
[00:02:06.705] ESP-ROM:esp32s3-20210327
[00:00:00.006] Build:Mar 27 2021
[00:00:00.000] rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
[00:00:00.005] SPIWP:0xee
[00:00:00.000] mode:DIO, clock div:1
[00:00:00.000] load:0x3fce3808,len:0x68
[00:00:00.005] load:0x403c9700,len:0x980
[00:00:00.000] load:0x403cc700,len:0x25a0
[00:00:00.000] SHA-256 comparison failed:
[00:00:00.005] Calculated: ba9b2d4eb238d33c8b2df1f10920a53b5a565dd3a0a23ff135232249c77b4a85
[00:00:00.005] Expected: bd6e2eb493ec8e2143cb0d39c23b5c2005f98ed268d8910021afbdfc58c4fde7
[00:00:00.005] Attempting to boot anyway...
[00:00:00.004] entry 0x403c98a4
```

My test code is:
```python
import alarm
import time
import board

board.DISPLAY.brightness = 1
time.sleep(1)

# Create a an alarm that will trigger 120 seconds from now.
time_alarm = alarm.time.TimeAlarm(monotonic_time=time.monotonic() + 120)
# Exit the program, and then deep sleep until the alarm wakes us.
print ("120 seconds in deep sleep")
board.DISPLAY.brightness = 0
alarm.exit_and_deep_sleep_until_alarms(time_alarm)
# Does not return, so we never get here.
```

tannewt, 2022-12-05:
Version: `Adafruit CircuitPython 8.0.0-beta.4-75-gd7874e65c on 2022-12-05; ESP32-S3-USB-OTG-N8 with ESP32S3`

sas42, 2022-12-06:
Version tested 12/6/2022 2:20 pm est
Adafruit CircuitPython 8.0.0-beta.4-77-g5cb867e85 on 2022-12-06; Adafruit Feather ESP32S3 No PSRAM with ESP32S3

This code does not even go into light sleep much less deep sleep

import alarm
import time
import board
import digitalio

# board.DISPLAY.brightness = 1

led = digitalio.DigitalInOut(board.LED)
led.switch_to_output(True)
time.sleep(2)
led.value = False

# Create a an alarm that will trigger 120 seconds from now.
time_alarm = alarm.time.TimeAlarm(monotonic_time=time.monotonic() + 30)
# Exit the program, and then deep sleep until the alarm wakes us.
print ("30 seconds in deep sleep")
# board.DISPLAY.brightness = 0

led.value = True
alarm.exit_and_deep_sleep_until_alarms(time_alarm)

BeatArnet, 2022-12-06:
Latest uf2-File installed - no change in behaviour: First time: 120 seconds sleep, from then on starts immediately

```py
#Adafruit` CircuitPython 8.0.0-beta.4-77-g5cb867e85 on 2022-12-06; Adafruit Feather ESP32-S3 TFT with ESP32S3
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

# Adafruit esp32-s3 TFT feather Board
# mit drei Sensoren:
# - Distanz vl53l4cd
# - Helligkeit bh1750
# - Luftqualität, Luftdruck, Temperatur, Feuchtigkeit, Höhe über Meer bme680
# Soll die Distanz zur Wasseroberfläche messen, aber dank der Sensorenabschaltung im DeepSleep auch über Tage messen können
# Beat Arnet, 01.10.2022
# art, 13.10.2022
#    Es fehlt noch der Schlafmodus und die Messung der Batteriespannung
# art, 15.10.2022
# Circuitpython 8.0.0 Beta 4 installiert und die beiden bisher nicht funktionierenden Funktionen getestet;
# - Batteriespannung messen: Funktioniert nur mit vorangehendem i2c-Scanning
# - deep sleep funktioniert nicht, auch wenn nur Batterie angehängt ist. der esp32 startet sofort wieder und macht keine Pause

import time
import alarm
import board
import digitalio
import adafruit_bme680
import adafruit_bh1750
import adafruit_vl53l4cd
from adafruit_lc709203f import LC709203F, PackSize
import displayio
import terminalio
from adafruit_display_text import label
import ssl
import socketpool
import wifi
import adafruit_minimqtt.adafruit_minimqtt as MQTT
import json
import gc

board.TFT_I2C_POWER
# TFT_I2C_POWER initialisieren
p = digitalio.DigitalInOut(board.TFT_I2C_POWER)
p.switch_to_output()   # turned the display OFF (default output is False)
p.value = True  # Display und Sensoren einschalten

# Create sensor object, communicating over the board's default I2C bus
i2c = board.I2C()  # uses board.SCL and board.SDA
bme680 = adafruit_bme680.Adafruit_BME680_I2C(i2c, debug=False)
bh1750 = adafruit_bh1750.BH1750(i2c)
vl53 = adafruit_vl53l4cd.VL53L4CD(i2c)

# TFT-Display initialisieren
# built-in display
display = board.DISPLAY

# Make the display context
main_group = displayio.Group()
display.show(main_group)

# change this to match the location's pressure (hPa) at sea level
bme680.sea_level_pressure = 995.0

# You will usually have to add an offset to account for the temperature of
# the sensor. This is usually around 5 degrees but varies by use. Use a
# separate temperature sensor to calibrate this one.
temperature_offset = -5

# OPTIONAL: can set non-default values
vl53.inter_measurement = 0
vl53.timing_budget = 200
[... 272 lines cut ...]
#del i2c
#del bme680
#del bh1750
#del vl53
#del display
#del battery_monitor
# garbage collector
#gc.collect()
# Powering down TFT-display and sensors
p.value = False 

# Create a an alarm that will trigger 10 seconds from now.
#time_alarm = alarm.time.TimeAlarm(monotonic_time=time.monotonic() + 4)
# Do a light sleep until the alarm wakes us.
#print ("4 Sekunden im leichtem Schlaf")
#alarm.light_sleep_until_alarms(time_alarm)
# Finished sleeping. Continue from here.

# Create a an alarm that will trigger 120 seconds from now.
time_alarm = alarm.time.TimeAlarm(monotonic_time=time.monotonic() + 120)
# Exit the program, and then deep sleep until the alarm wakes us.
print ("120 Sekunden im tiefen Schlaf")
alarm.exit_and_deep_sleep_until_alarms(time_alarm)
# Does not return, so we never get here.
```

Neradoc, 2022-12-08:
In my tests with beta 5, it never goes to deep sleep but just resets, like before I think.
Feather ESP32-S3 TFT
```
Adafruit CircuitPython 8.0.0-beta.5 on 2022-12-08; Adafruit Feather ESP32-S3 TFT with ESP32S3
```
Running that code.
```py
import alarm
import time
time_alarm = alarm.time.TimeAlarm(monotonic_time=time.monotonic() + 20)
alarm.exit_and_deep_sleep_until_alarms(time_alarm)
```
Does not go to deep sleep but resets immediately instead when on power only (USB power adapter or battery).
It does "sleep" for approximately 20 seconds when on USB (fake deep sleep).

tannewt, 2022-12-09:
I tested this yesterday and found that the ULP was incorrectly waking up the S3. I'll try and sort it out today.

PR #7359 (2022-12-21): Rework the coproc API
It is now a generic `memorymap` API and an ESP specific `espulp` module.

Fixes #7218. Fixes #3234. Fixes #7300.
commit f8ac1d92 (2022-12-21): Rework the coproc API

It is now a generic `memorymap` API and an ESP specific `espulp` module.

Fixes #7218. Fixes #3234. Fixes #7300.
PR #7441 (2023-01-11): Ignore ESP-IDF wakeup cause when light sleeping
The IDF's wakeup cause is only for deep sleep. Without ignoring it, light sleep will wake up too early when done after a deep sleep wake.

Fixes #7300
commit fbd5f273 (2023-01-11): Ignore ESP-IDF wakeup cause when light sleeping

The IDF's wakeup cause is only for deep sleep. Without ignoring it,
light sleep will wake up too early when done after a deep sleep
wake.

Fixes #7300
closed 2022-12-21 by dhalbert