adafruit/circuitpython #7301: Random save delay when using Thonny

LovelyA72, 2022-12-03:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.4 on 2022-10-30; Adafruit Pybadge with samd51j19
```

### Code/REPL

```python
anything
```

### Behavior

Press ctrl+s in Thonny to save takes a long time with error message on display
[image]

### Additional information

Thonny issue but no response: https://github.com/thonny/thonny/issues/2561

dhalbert, 2022-12-04:
The filesystem might be corrupted, causing it to be read-only. Retrieve off the files you need, then do:
```py
import storage
storage.erase_filesystem()
```
and put the files back. See if that fixes it.

LovelyA72, 2022-12-05:
After erasing the file system on a pi pico W the issue persists. Not tested on EdgeBadge yet but seems like it's unlikely due to a file system corruption. How do you get logs like what you see on EdgeBadge's screen on a Pico W?

dhalbert, 2022-12-05:
If Thonny is trying to write the file by executing REPL commands, which it does no boards with no mass-storage USB connection, then you will see that message because the filesystem is read-only to the CircuitPython side. It is read-write to the the host computer, which has a regular USB-drive connection.

I just tried to write a file with an ESP32-S2 board, which presents CIRCUITPY as a USB drive. Thonny is able to write. I think Thonny tries to figure out that CIRCUITPY is present and write directly. I don't know why it's failing to do that on the Pico W. It may be a setting. You might want to ask this question on the Thonny mailing list or other Thonny support place.

If you try to write a file not using Thonny, but with some other editor, does it work?

LovelyA72, 2022-12-05:
Yes, it does work, the usb drive mode works fine. I think something is wrong with thonny's REPL.

dhalbert, 2022-12-05:
MicroPython allows writes from the Python code and the host, which can cause corruption if the writes are concurrent.

LovelyA72, 2022-12-05:
Closed until there's more evidence it's a CircuitPython issue

closed 2022-12-05 by LovelyA72

dhalbert, 2022-12-05:
Working with @LovelyA72 in discord, I could not reproduce this problem in Windows 10 using both the standalone install and the pip3 version of Thonny 4.0.1.

LovelyA72, 2023-01-26:
This is indeed not a CircuitPython issue. Instead it's a Thonny issue. Seems like it is trying to access disconnected network mapped drives. Which is reasonable since CircuitPython also uses mass storage device for accessing codes.