adafruit/circuitpython #7304: Using two SPI devices: https://www.adafruit.com/product/5300

horsto, 2022-12-04:
### CircuitPython version

```python
CircuitPython 7.3.3 on 2022-08-29; Adafruit Feather ESP32-S2 TFT with ESP32S2
```

### Code/REPL

```python
import board
import busio
import digitalio
import adafruit_requests as requests
from adafruit_wiznet5k.adafruit_wiznet5k import WIZNET5K
import adafruit_wiznet5k.adafruit_wiznet5k_socket as socket

print("Wiznet5k WebClient Test")

TEXT_URL = "http://wifitest.adafruit.com/testwifi/index.html"
JSON_URL = "http://api.coindesk.com/v1/bpi/currentprice/USD.json"

# For Adafruit Ethernet FeatherWing
cs = digitalio.DigitalInOut(board.D10)
# For Particle Ethernet FeatherWing
# cs = digitalio.DigitalInOut(board.D5)
spi_bus = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
```

### Behavior

```python
code.py output:
Wiznet5k WebClient Test
Traceback (most recent call last):
  File "code.py", line 20, in <module>
ValueError: SCK in use

```

### Description

I am using the [ethernet shield](https://www.adafruit.com/product/3201) together with the Feather [ESP32-S2 TFT](https://www.adafruit.com/product/5300). It seems that the ESP32-S2 Feather TFT display is using the SPI clock before the ethernet shield has a chance to establish communication. How can the two devices (internal TFT and ethernet shield) work at the same time?

anecdata, 2022-12-04:
Did you try `spi_bus = board.SPI()`?

horsto, 2022-12-04:
Thanks, that works! 
At least I can work with the ethernet featherwing now. 
But how can I then use the display in parallel?

anecdata, 2022-12-04:
Should be no trouble, the display is already set up and you can use it for REPL / terminal or `displayio` in code.py

btw, Discord is a great place for support-type questions... https://adafru.it/discord

horsto, 2022-12-04:
Thanks a lot - and I will write future support-type questions on discord.

closed 2022-12-04 by horsto