adafruit/circuitpython #7305: Crash when extending PWMOut class

ali1234, 2022-12-05:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.4 on 2022-10-30; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
import board
import pwmio

class PWMOutExtend(pwmio.PWMOut):
    def __init__(self, pin, frequency=1000, slew_rate=1000):
        super().__init__(pin, frequency=frequency)
        self.foo_bar_baz = 0
        self.duty_cycle = 0

led = PWMOutExtend(board.GP25, frequency=1000)

print('before')
led.foo_bar_baz = 0
print('after')
```

### Behavior

You see "before" on the REPL and then the device freezes. If you put it in code.py you now have to enter safe mode to access CIRCUITPY.

### Description

Every line of the example is necessary except for the prints. In particular all three assignments are necessary.

I can't reproduce it when extending a class I defined myself.

tannewt, 2022-12-05:
Generally native classes aren't subclassable because our C code doesn't always check to ensure it was given the native struct representation. Ideally we'd have a way to do this for everything.

In the meantime, I'd suggest wrapping PWMOut instead of subclassing.

dhalbert, 2024-01-03:
It would be nice if we could forbid or warn when a native class that can't handle subclassing is subclassed. Classes that do handle this properly could be marked as such. This is probably a change to make upstream.

TonyLHansen, 2024-01-03:
Such an error message would be appreciated.