adafruit/circuitpython #7308: Circuitpython 7.x issue on Adafruit Feather ESP32-S2

k1w1tim, 2022-12-06:
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.

[boot_out.txt](https://github.com/adafruit/circuitpython/files/10163048/boot_out.txt)
[code.txt](https://github.com/adafruit/circuitpython/files/10163085/code.txt)

I tried uploading the contents of the Circuitpython drive on my windows10 machine but as the message below states it won't accept them??? what am I doing wrong? boot_out.txt is the only file I can attach. Code.txt is my code.py file.

k1w1tim, 2022-12-06:
Also one change to code.txt is that I was adapting the gfx_ili9341_test.py code in the examples for the gfx library and so the baudrate was set to 32000000 when it broke.
i.e. "BAUDRATE=24000000" was "BAUDRATE=32000000". 

Possible cause?

tannewt, 2022-12-06:
Please try the "absolute newest" version of CircuitPython too. It is possible the bug has already been fixed in newer versions.

k1w1tim, 2022-12-06:
Ok that got it thanks, much appreciated.

Also while I have you, where do I find documentation & downloads for the "machine" library or am I not meant to use it with this board?

closed 2022-12-07 by tannewt