adafruit/circuitpython #7310: ANT and ANT+ on NRF52 boards

casainho, 2022-12-06:
I wish to be able to use ANT and ANT+ on NRF52 boards. I use it for implementing ANT+LEV EBike standard as also GPS cycling display Garmin Edge page change but in C, similar to Nordic SDK code, if anyone is interested: https://github.com/OpenSourceEBike/EV_Display_Bluetooth_Ant/tree/main/firmware/display/antplus_lev

dhalbert, 2022-12-06:
We have not supported ANT and ANT+ because of the licensing fee requirements:
https://www.thisisant.com/developer/components/view-all-components/nrf52-ant-stack-licensing-faq

When I was researching this a while ago, I found that some bike devices that support ANT+ also support BLE, such as the Wahoo sensors.

dhalbert, 2022-12-07:
It is not completely clear to me if Adafruit would be required to pay the licensing fee, because we are selling boards that could load the ANT+-capable software, even if just for hobbyist use, or if the licensing fee is meant to apply only to commercial products that use ANT+.

casainho, 2022-12-07:
Yes, the DIY EBike display I built have simultaneous Bluetooth and ANT+, it works very well: https://opensourceebike.github.io

But I would prefer to develop again that display based in Python code instead. And ANT+ is a MUST for cycling devices, from electronics gears shifters, tire pressure sensors and the EBike motors.

[image]

[image]

[image]

cujomalainey, 2023-07-17:
FYI @orrmany wrote this awesome article on how to load some adafruit devices https://blogarak.wordpress.com/2020/03/15/s340-softdevice-adafruit-nrf52840-express-feather/