adafruit/circuitpython #7316: CIRCUITPY mount permissions wrong when booting to console

mattbrailsford, 2022-12-08:
### CircuitPython version

```python
7.3.3
```

### Code/REPL

```python
n/a
```

### Behavior

n/a

### Description

When connecting a Tiny 2040 with circuitpython installed to a Raspberry PI 2 Zero W I'm seeing issues depending on whether I boot to desktop or boot to console.

If I boot to desktop with autologin then my circuitpython drive mounts successfully and the code on it runs, but if I set the pi to boot to console (either with or without autologin) the CIRCUITPY drive mounts with insufiscient permissions and fails to run the code on the device.

You can see in this screenshot that when booting to desktop, the mount owner is `pi` and the drive has the correct permissions, but when I boot to console, the owner becomes `root` and I no longer have permissions on the mount and so it is inaccessible and the code doesn't run.

[image]

dhalbert, 2022-12-08:
This seems like a user group issue, or some automount difference. I don't think it has to do with CircuitPython. I would think you would see the same issues with any USB mass storage device, such as a USB stick. Try it with a regular USB stick.

dhalbert, 2023-01-09:
Closing after a month. Please comment if your problem persists.

closed 2023-01-09 by dhalbert