adafruit/circuitpython #7319: Builds for FOMU failed to work properly after 7.0.0-beta.0

qqice, 2022-12-08:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0-beta.0 on 2021-08-24; Fomu with VexRiscv
```

### Code/REPL

```python
every code
```

### Behavior

Flashing every firmware after 7.0.0-beta.0 (from 7.0.0-rc.0) using dfu-util will cause circuitpython failed to work properly. A drive named CIRCUITPY won't appeared. Though there is a serial port appeared in the Device Manager, every attempt to connect to the board (for example, using Thonny) through this port will end up in an error.

qqice, 2022-12-08:
I think the commit b14b2945167fd20c313e3bf8e1337b73355dc7ff may have broken something so the board can't function properly. Please check with it.

qqice, 2022-12-08:
But I think the number of people holding this development board is quite small. Otherwise this problem shouldn't have been discovered after 14 months.

tannewt, 2026-10-13:
We no longer support the litex port.

closed 2026-10-13 by tannewt