adafruit/circuitpython #7326: mDNS find yields duplicate entries if another device is also find-ing mDNS (& other oddities)

anecdata, 2022-12-10:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.5 on 2022-12-08; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
# mDNS finder

import time
import wifi
import mdns
import microcontroller
from secrets import secrets

MDNSFINDTIMEOUT = 5

def connect():
    # to connect if no web workflow, or to reconnect if disconnexted
    if not wifi.radio.ipv4_address:
        print (f"Connecting to wifi AP..", end="")
        while True:
            try:
                print (f".", end="")
                wifi.radio.connect(secrets["ssid"], secrets["password"])
                time.sleep(0.100)  # delay sometimes needed for RP2
                break
            except ConnectionError as e:
                pass  # ConnectionError: Authentication failure
        print (f" IPv4={wifi.radio.ipv4_address}")

time.sleep(2)  # wait for serial
print(f"{'='*25}\nStarting mDNS server...")
try:
    m = mdns.Server(wifi.radio)
except RuntimeError as e:
    if "mDNS already initialized" in repr(e):
        print(f"RuntimeError: {e}. Resetting...")
        microcontroller.reset()
    else:
        print(f"RuntimeError: {e}")

while True:
    connect()
    print(f"{'-'*25}\nFinding mDNS hosts from {wifi.radio.ipv4_address}...")
    for service in m.find(service_type="_circuitpython", protocol="_tcp", timeout=MDNSFINDTIMEOUT):
        print(f"{service.instance_name} {service.hostname} {service.ipv4_address}:{service.port}")
    time.sleep(5)
```

### Behavior

Note that the above code won't work on `raspberrypi` if web workflow is enabled by an `.env` file.

The main issue behavior is that when no other devices are `find`-ing mDNS on the network, `raspberrypi` will yield a set of unique entries for services found. But if another device is `find`-ing mDNS (for example, an `espressif` device running the same code), `raspberrypi` will yield duplicate entries in the services found.

On `raspberrypi`, mDNS is not shutdown on reload, so `m = mdns.Server(wifi.radio)` gets `RuntimeError: mDNS already initialized`. Even on reset, with web workflow enabled, `m = mdns.Server(wifi.radio)` gets `RuntimeError: mDNS already initialized`.

`espressif` shuts down mDNS on reload, so after a reload, `m = mdns.Server(wifi.radio)` works. It works with web workflow enabled too.

Side note: there is a vestigial debug print in the `find` results also: `found service 0x20******`, where ****** increases on successive results. These numbers _are_ unique even when the instance name, hostname, and IPv4 are duplicated in the results.

anecdata, 2022-12-21:
Opening a separate issue for this:

> On `raspberrypi`, mDNS is not shutdown on reload, so `m = mdns.Server(wifi.radio)` gets `RuntimeError: mDNS already initialized`. Even on reset, with web workflow enabled, `m = mdns.Server(wifi.radio)` gets `RuntimeError: mDNS already initialized`. **This is probably the larger issue, but I'm not entirely sure how this should work across ports.**

> `espressif` shuts down mDNS on reload, so after a reload, `m = mdns.Server(wifi.radio)` works. It works with web workflow enabled too.

tannewt, 2023-04-14:
Would updating the documentation be enough to fix this? I'd rather not have to track all returned values to filter duplicates.

anecdata, 2023-04-19:
Still occurs with:
`Adafruit CircuitPython 8.1.0-beta.1-25-g4d7e54310 on 2023-04-16; Raspberry Pi Pico W with rp2040`
It probably isn't worth the extra code to filter, easy enough to find uniques in Python. I thought it might arise from something in the driver or sdk.

jepler, 2023-06-21:
closing as it looks like it's probably an issue in lwip's mdns implementation, and there are workarounds.

closed 2023-06-21 by jepler