adafruit/circuitpython #7328: CI builds failing in `tj-actions/changed-files`

dhalbert, 2022-12-11:
In the past couple of days, PR's are failing due to errors that seem to be caused by `tj-actions/changed-files`.
#7132 introduced the use of that action.

Here is a typical failure, from https://github.com/adafruit/circuitpython/actions/runs/3666904634/jobs/6198998390, with the relevant log section extracted:
```
Run tj-actions/changed-files@v34
  with:
    json: true
    separator:  
    include_all_old_new_renamed_files: false
    old_new_separator: ,
    old_new_files_separator:  
    files_separator: 
  
    files_ignore_separator: 
  
    path: .
    quotepath: true
    dir_names: false
    fetch_depth: 40
    max_fetch_depth: 5000
    since_last_remote_commit: false
  env:
    pythonLocation: /opt/hostedtoolcache/Python/3.11.0/x64
    PKG_CONFIG_PATH: /opt/hostedtoolcache/Python/3.11.0/x64/lib/pkgconfig
    Python_ROOT_DIR: /opt/hostedtoolcache/Python/3.11.0/x64
    Python[2](https://github.com/adafruit/circuitpython/actions/runs/3666904634/jobs/6198998390#step:29:2)_ROOT_DIR: /opt/hostedtoolcache/Python/[3](https://github.com/adafruit/circuitpython/actions/runs/3666904634/jobs/6198998390#step:29:3).11.0/x6[4](https://github.com/adafruit/circuitpython/actions/runs/3666904634/jobs/6198998390#step:29:4)
    Python3_ROOT_DIR: /opt/hostedtoolcache/Python/3.11.0/x64
    LD_LIBRARY_PATH: /opt/hostedtoolcache/Python/3.11.0/x64/lib
    CP_VERSION: 8.0.0-beta.[5](https://github.com/adafruit/circuitpython/actions/runs/3666904634/jobs/6198998390#step:29:5)-5-ge88f8d[6](https://github.com/adafruit/circuitpython/actions/runs/3666904634/jobs/6198998390#step:29:6)5[7](https://github.com/adafruit/circuitpython/actions/runs/3666904634/jobs/6198998390#step:29:7)
Run # "Calculating the previous and current SHA..."
  # "Calculating the previous and current SHA..."
  bash $GITHUB_ACTION_PATH/diff-sha.sh
  shell: /usr/bin/bash --noprofile --norc -e -o pipefail {0}
  env:
    pythonLocation: /opt/hostedtoolcache/Python/3.11.0/x64
    PKG_CONFIG_PATH: /opt/hostedtoolcache/Python/3.11.0/x64/lib/pkgconfig
    Python_ROOT_DIR: /opt/hostedtoolcache/Python/3.11.0/x64
    Python2_ROOT_DIR: /opt/hostedtoolcache/Python/3.11.0/x64
    Python3_ROOT_DIR: /opt/hostedtoolcache/Python/3.11.0/x64
    LD_LIBRARY_PATH: /opt/hostedtoolcache/Python/3.11.0/x64/lib
    CP_VERSION: [8](https://github.com/adafruit/circuitpython/actions/runs/3666904634/jobs/6198998390#step:29:8).0.0-beta.5-5-ge88f8d657
    GITHUB_SERVER_URL: https://github.com
    GITHUB_REPOSITORY: adafruit/circuitpython
    GITHUB_REF: refs/pull/7327/merge
    GITHUB_BASE_REF: main
    GITHUB_HEAD_REF: in-group-remove
    GITHUB_WORKSPACE: /home/runner/work/circuitpython/circuitpython
    GITHUB_EVENT_HEAD_REPO_FORK: true
    GITHUB_EVENT_PULL_REQUEST_BASE_SHA: [9](https://github.com/adafruit/circuitpython/actions/runs/3666904634/jobs/6198998390#step:29:9)06cd6eb65726a9ca04974582344d772a04929df
    GITHUB_EVENT_PULL_REQUEST_HEAD_SHA: 64[12](https://github.com/adafruit/circuitpython/actions/runs/3666904634/jobs/6198998390#step:29:12)d971facb6c0c75aaced897f39dc0a7d[17](https://github.com/adafruit/circuitpython/actions/runs/3666904634/jobs/6198998390#step:29:17)bb2
    GITHUB_EVENT_BEFORE: bd91959ce3eb68375739680bd795fe3f7d0973f0
    GITHUB_EVENT_FORCED: 
    GITHUB_REFNAME: 7327/merge
    INPUT_SHA: 
    INPUT_BASE_SHA: 
    INPUT_SINCE: 
    INPUT_UNTIL: 
    INPUT_PATH: .
    INPUT_FETCH_DEPTH: 40
    INPUT_MAX_FETCH_DEPTH: 5000
    INPUT_SINCE_LAST_REMOTE_COMMIT: false
changed-files-diff-sha
  Verifying git version...
  Valid git version found: (2.38.1)
  Running on a pull request event...
  Fetching remote refs...
  remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0        
  From https://github.com/adafruit/circuitpython
   * [new branch]          main       -> origin/main
  branch 'main' set up to track 'origin/main'.
  Fetching 40 commits...
  fatal: error in object: unshallow 98002cf0abe5019cf3c57e6c[18](https://github.com/adafruit/circuitpython/actions/runs/3666904634/jobs/6198998390#step:29:18)96c4c5dd[22](https://github.com/adafruit/circuitpython/actions/runs/3666904634/jobs/6198998390#step:29:22)6f[24](https://github.com/adafruit/circuitpython/actions/runs/3666904634/jobs/6198998390#step:29:24)
  Error: Process completed with exit code 1[28](https://github.com/adafruit/circuitpython/actions/runs/3666904634/jobs/6198998390#step:29:29).
```
Note the `INPUT_FETCH_DEPTH: 40` setting, which is suspiciously the same as the `Fetching 40 commits..` just before the fatal error.

I don't really understand the reason for the error, but I wonder if it is because we are doing shallow clones of `circuitpython`, with `depth=1`. Or perhaps we just need to tweak the `tj-actions/changed-files` parameters.

_*If*_ it is the case that the `depth=1` clones are the issue, here is some more discussion:

Shallow clones are missing information. In #7225, I suggested we might use partial clones instead. They do lazy fetching of a tree, but from a user's point of view they are "complete", fetching more info dynamically as necessary.

There are "treeless" partial clones and "blobless" partial clones. Both are a little slower (2 seconds vs 5 seconds to fetch `circuitpython` for me), but that's not too bad. It's not clear to me which is better for a "fetch, build, and discard" CI job, but there is some discussion here:

https://github.blog/2020-12-21-get-up-to-speed-with-partial-clone-and-shallow-clone/  (detailed overview)
https://gist.github.com/stormwild/b7ae53f401ecf9f51f097a43d01be23b (seems to recommend blobless for CI builds)
I'm sure there are more things to read too.

closed 2022-12-11 by dhalbert

MicroDev1, 2022-12-11:
We have a new failure now: (See: https://github.com/adafruit/circuitpython/actions/runs/3669965941/jobs/6204173152)
```python
  Error: Similar commit hashes detected: previous sha: 66eca9c35e9da27232c67b9504b0bec6b6828dd0 is equivalent to the current sha: 66eca9c35e9da27232c67b9504b0bec6b6828dd0.
  Error: Please verify that both commits are valid, and increase the fetch_depth to a number higher than 40.
  Error: Process completed with exit code 1.
```

dhalbert, 2022-12-11:
@jepler want to try upping the fetch depth as a new commit in #7321, as was done in https://github.com/adafruit/circuitpython/pull/7329/commits/628865b2357bd1c344757a55ffbed763dc506121 ?

dhalbert, 2022-12-11:
or @MicroDev1 you can try that. I am just about to go out for a while.

MicroDev1, 2022-12-11:
Here is the diff of `tj-actions/changed-files` logs in PR 7324 `Good-CI` and PR 7327 `Bad-CI`:
```patch
diff --git ci.log ci.log
index 46903bd36..5cf9191be 100644
--- ci.log
+++ ci.log
@@ -1,131 +1,69 @@
 Run tj-actions/changed-files@v34
   with:
     json: true
     separator:
     include_all_old_new_renamed_files: false
     old_new_separator: ,
     old_new_files_separator:
     files_separator:

     files_ignore_separator:

     path: .
     quotepath: true
     dir_names: false
     fetch_depth: 40
     max_fetch_depth: 5000
     since_last_remote_commit: false
   env:
     pythonLocation: /opt/hostedtoolcache/Python/3.11.0/x64
     PKG_CONFIG_PATH: /opt/hostedtoolcache/Python/3.11.0/x64/lib/pkgconfig
     Python_ROOT_DIR: /opt/hostedtoolcache/Python/3.11.0/x64
     Python2_ROOT_DIR: /opt/hostedtoolcache/Python/3.11.0/x64
     Python3_ROOT_DIR: /opt/hostedtoolcache/Python/3.11.0/x64
     LD_LIBRARY_PATH: /opt/hostedtoolcache/Python/3.11.0/x64/lib
-    CP_VERSION: 8.0.0-beta.5-4-g3a15be7d3
+    CP_VERSION: 8.0.0-beta.5-5-ge88f8d657
 Run # "Calculating the previous and current SHA..."
   # "Calculating the previous and current SHA..."
   bash $GITHUB_ACTION_PATH/diff-sha.sh
   shell: /usr/bin/bash --noprofile --norc -e -o pipefail {0}
   env:
     pythonLocation: /opt/hostedtoolcache/Python/3.11.0/x64
     PKG_CONFIG_PATH: /opt/hostedtoolcache/Python/3.11.0/x64/lib/pkgconfig
     Python_ROOT_DIR: /opt/hostedtoolcache/Python/3.11.0/x64
     Python2_ROOT_DIR: /opt/hostedtoolcache/Python/3.11.0/x64
     Python3_ROOT_DIR: /opt/hostedtoolcache/Python/3.11.0/x64
     LD_LIBRARY_PATH: /opt/hostedtoolcache/Python/3.11.0/x64/lib
-    CP_VERSION: 8.0.0-beta.5-4-g3a15be7d3
+    CP_VERSION: 8.0.0-beta.5-5-ge88f8d657
     GITHUB_SERVER_URL: https://github.com
     GITHUB_REPOSITORY: adafruit/circuitpython
-    GITHUB_REF: refs/pull/7324/merge
+    GITHUB_REF: refs/pull/7327/merge
     GITHUB_BASE_REF: main
-    GITHUB_HEAD_REF: doc-remove-asyncio-update-library-pages
+    GITHUB_HEAD_REF: in-group-remove
     GITHUB_WORKSPACE: /home/runner/work/circuitpython/circuitpython
-    GITHUB_HEAD_REPO_FORK: true
-    GITHUB_EVENT_PULL_REQUEST_BASE_SHA: 4fa0c4d10822a316da7b374bee39ed3591056acb
-    GITHUB_EVENT_BEFORE:
+    GITHUB_EVENT_HEAD_REPO_FORK: true
+    GITHUB_EVENT_PULL_REQUEST_BASE_SHA: 906cd6eb65726a9ca04974582344d772a04929df
+    GITHUB_EVENT_PULL_REQUEST_HEAD_SHA: 6412d971facb6c0c75aaced897f39dc0a7d17bb2
+    GITHUB_EVENT_BEFORE: bd91959ce3eb68375739680bd795fe3f7d0973f0
     GITHUB_EVENT_FORCED:
-    GITHUB_REFNAME: 7324/merge
+    GITHUB_REFNAME: 7327/merge
     INPUT_SHA:
     INPUT_BASE_SHA:
     INPUT_SINCE:
     INPUT_UNTIL:
     INPUT_PATH: .
     INPUT_FETCH_DEPTH: 40
[... 56 lines cut ...]
-    GITHUB_WORKSPACE: /home/runner/work/circuitpython/circuitpython
-    GITHUB_BASE_REF: main
-    INPUT_FILES_PATTERN_FILE: /tmp/01c0a6cb-9d26-4d50-b17c-5531970820d7.txt
-    INPUT_SEPARATOR:
-    INPUT_PATH: .
-    INPUT_PREVIOUS_SHA: 4fa0c4d10822a316da7b374bee39ed3591056acb
-    INPUT_CURRENT_SHA: fa96bcce84b0ac6feb34b49d55c05362ea7269d4
-    INPUT_TARGET_BRANCH: main
-    INPUT_CURRENT_BRANCH: doc-remove-asyncio-update-library-pages
-    INPUT_QUOTEPATH: true
-    INPUT_INCLUDE_ALL_OLD_NEW_RENAMED_FILES: false
-    INPUT_OLD_NEW_SEPARATOR: ,
-    INPUT_OLD_NEW_FILES_SEPARATOR:
-    INPUT_DIFF_RELATIVE:
-    INPUT_DIR_NAMES: false
-    INPUT_DIR_NAMES_MAX_DEPTH:
-    INPUT_JSON: true
-    INPUT_HAS_CUSTOM_PATTERNS: false
-changed-files
-  Resolving repository path: /home/runner/work/circuitpython/circuitpython/.
-  Retrieving changes between 4fa0c4d10822a316da7b374bee39ed3591056acb (main) → fa96bcce84b0ac6feb34b49d55c05362ea7269d4 (doc-remove-asyncio-update-library-pages)
+  Fetching 40 commits...
+  fatal: error in object: unshallow fc975d61478fdb7beb6084c35458351015ba96df
+  Error: Process completed with exit code 128.
```

hamishwillee, 2023-05-06:
I'm still getting this with latest release, when using `pull_request_target`: https://github.com/PX4/PX4-user_guide/blob/main/.github/workflows/flaw_checker.yml
- Failed build here: https://github.com/PX4/PX4-user_guide/actions/runs/4901190561/jobs/8752318407?pr=2487

It is quite likely something I am doing wrong'; this is for anyone else running into the issue.

GuyAfik, 2023-05-21:
We are having the same [issue](https://github.com/demisto/content/actions/runs/5038637469/jobs/9036707744) when using tj-actions/changed-files@v35.1.0, any idea here? how can we fix it?

PR #7329 (2022-12-11): Temporary Fix for #7328 
Try and fix #7328.

1. Update `tj-actions/changed-files` to `v34.5.3`.
2. Use `fetch-depth: 0` to avoid shallow clone issue.