adafruit/circuitpython #7330: Bluetooth is not visible on the MakerDiary nrf52840

idreamsi, 2022-12-11:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.3: MakerDiary nrf52840 USB dongle
```

### Code/REPL

```python
import time
import random
from ble_json_service import SensorService
from adafruit_ble import BLERadio
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement

# Create BLE radio, custom service, and advertisement.
ble = BLERadio()
service = SensorService()
advertisement = ProvideServicesAdvertisement(service)

# Function to get some fake weather sensor readings for this example in the desired unit.
def measure(unit):
    temperature = random.uniform(0.0, 10.0)
    humidity = random.uniform(0.0, 100.0)
    if unit == "fahrenheit":
        temperature = (temperature * 9.0 / 5.0) + 32.0
    return {"temperature": temperature, "humidity": humidity}

# Advertise until another device connects, when a device connects, provide sensor data.
while True:
    print("Advertise services")
    ble.stop_advertising()  # you need to do this to stop any persistent old advertisement
    ble.start_advertising(advertisement)

    print("Waiting for connection...")
    while not ble.connected:
        pass

    print("Connected")
    while ble.connected:
        settings = service.settings
        measurement = measure(settings.get("unit", "celsius"))
        service.sensors = measurement
        print("Settings: ", settings)
        print("Sensors: ", measurement)
        time.sleep(0.25)

    print("Disconnected")
```

### Behavior

There are no errors and it compiles without problems.

### Description

I have two different boards with nrf50840 [XIAO BLEsense and MakerDiary nrf52840 usb dongle]. The code works fine on XIAO BLE and the device is visible in the **nRF Connect** app on mobile phone, but when I put the [same code](https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/main/examples/ble_json_peripheral.py) on MakerDiary without getting an error, no device is visible in the **nRF Connect** search section!

dhalbert, 2022-12-11:
There are multiple MakerDiary boards. Does yours match the photograph on https://circuitpython.org for your board or dongle?

We have not had similar reports, so I'm wondering about whether there might be a hardware problem with yours. Do you have non-CircuitPython software that is successfully doing BLE?

idreamsi, 2022-12-12:
> There are multiple MakerDiary boards. Does yours match the photograph on https://circuitpython.org for your board or dongle?

[This is my board](https://makerdiary.com/products/nrf52840-mdk-usb-dongle) and I use [this link](https://circuitpython.org/board/makerdiary_nrf52840_mdk_usb_dongle/).

> We have not had similar reports, so I'm wondering about whether there might be a hardware problem with yours. Do you 
have non-CircuitPython software that is successfully doing BLE?

Bluetooth board has no problem and is healthy, Because it works correctly with Arduino example.

dhalbert, 2023-10-25:
Closing due to inactivity. Please reopen if this is still an issue for you.

closed 2023-10-25 by dhalbert