adafruit/circuitpython #7334: revamp supervisor allocation API

jepler, 2022-12-12:
I have a sketch at https://github.com/jepler/circuitpython/commit/e635457a32b1cc3b4b611808fcdd16951d12f99c but it doesn't boot.

The idea is to do more to leverage the gc allocator to make the supervisor allocator work.

See the commit message for more details.

Entered as an issue at @tannewt 's request

I don't plan to work on this until after 8 is out, but it was in my brain and I had to do at least this much of it.

tannewt, 2022-12-13:
So this always has the heap inited on allocation but then it tries to move movable allocations on init? Is there a period where it is deinited?

jepler, 2023-10-03:
This may be mooted by the split heap code we'll take from micropython.

Yes, in this work there's a moment where the heap is deinited, but "supervisor allocations" are still live. Immediately after the heap is reinited, the supervisor allocations are then marked in use by magic / layer violation.

PR #8553 (2023-11-08): Switch all ports to auto-growing split heap
This simplifies allocating outside of the VM because the VM doesn't take up all remaining memory by default.

On ESP we delegate to the IDF for allocations. For all other ports, we use TLSF to manage an outer "port" heap. The IDF uses TLSF internally and we use their fork for the other ports.

This also removes the dynamic C stack sizing. It wasn't often used and is not possible with a fixed outer heap.

Fixes #8512. Fixes #7334.
commit 8137e2d6 (2023-11-08): Switch all ports to auto-growing split heap

This simplifies allocating outside of the VM because the VM doesn't
take up all remaining memory by default.

On ESP we delegate to the IDF for allocations. For all other ports,
we use TLSF to manage an outer "port" heap. The IDF uses TLSF
internally and we use their fork for the other ports.

This also removes the dynamic C stack sizing. It wasn't often used
and is not possible with a fixed outer heap.

Fixes #8512. Fixes #7334.
closed 2023-11-08 by dhalbert