adafruit/circuitpython #7340: esp32_camera: creating a camera object makes the displayio stop working

deshipu, 2022-12-13:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.5 on 2022-12-08; S2Mini with ESP32S2-S2FN4R2
```

### Code/REPL

```python
import board
import displayio
import adafruit_ili9341
import busio
import digitalio
import esp32_camera

displayio.release_displays()
display_bus = displayio.FourWire(
    busio.SPI(clock=board.IO7, MOSI=board.IO5),
    command=board.IO9,
    chip_select=board.IO3,
    reset=board.IO10,
    baudrate=40_000_000,
)
display = adafruit_ili9341.ILI9341(
    display_bus,
    width=320,
    height=240,
)

i2c = busio.I2C(scl=board.IO35, sda=board.IO33)
data_pins = (
    board.IO21, board.IO17, board.IO16, board.IO18,
    board.IO37, board.IO34, board.IO36, board.IO39,
)
c = esp32_camera.Camera(
    data_pins=data_pins,
    external_clock_pin=board.IO11,
    pixel_clock_pin=board.IO13,
    vsync_pin=board.IO40,
    href_pin=board.IO38,
    pixel_format=esp32_camera.PixelFormat.RGB565,
    frame_size=esp32_camera.FrameSize.QVGA,
    i2c=i2c,
    external_clock_frequency=16_000_000,
)
```

### Behavior

After running the above code, the display becomes white and stays white after the program finishes, never getting back to displaying the REPL. Commenting out the last line and doing a hard reset makes the display show REPL normally.

deshipu, 2022-12-13:
I'm sorry, I'm an idiot, false alarm.

closed 2022-12-13 by deshipu

algorni, 2023-01-11:
where are you running this code?

I'm using an ESP32-CAM with the ESP-EYE firmware: https://circuitpython.org/board/espressif_esp32_eye/

But i'm getting error while using such IO ports: AttributeError: 'module' object has no attribute 'IO35'