adafruit/circuitpython #7351: import not following its own rules, finds alternative w/same name

schaefer01, 2022-12-16:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.3 on 2022-08-29; Adafruit Feather RP2040 with rp2040
```

### Code/REPL

```python
from adafruit_wiznet5k.adafruit_wiznet5k import WIZNET5K
```

### Behavior

  File "code.py", line 5, in <module>
  File "adafruit_wiznet5k.py", line 51, in <module>
ImportError: no module named 'adafruit_wiznet5k.adafruit_wiznet5k_dhcp'

### Description

The initial context:
The import worked when there was only the wiznet library to contend with.
The code ran to completion but raised a timeout error because the mac address had not been set and the default mac address was not recognized by the firewall. 

The change that cause the import error:
I needed to debug the source to figure out what was going on, so I copied the wiznet source files (from github) into the top directory, and to baseline, I reran code.py as-is, expecting the run to completion with error as before. However, I got the error reported above, which would  only make sense if the import imported the top level rather than take the adafruit directory path as specified by the import statement.
So, my belief is that the import isn't following the rules and picking what it finds close by before following the import path in the import statement.
Once I renamed the source files, the import could not "see" them, and the original error returned.

### Additional information

Changing the file names (and all the import statements in the source) allows me to continue debugging, but the import path "following the rules" problem remains.

dhalbert, 2022-12-16:
`sys.path` is `['', '/', '/lib']`. So the files you copied into the top-level directory, CIRCUITPY, would be found first. But if you didn't copy the whole tree, then once the package started importing things inside itself, it wouldn't find things. Is the top-level package tree as complete as what is in `lib/`?

schaefer01, 2022-12-16:
hi,

  Good to know. I'll be careful of this in the future.

  Once I changed the source file names there were no import name collisions.

  The problem with the "featherwing" is there is no such thing as "ls -R", the whole OS seems
   to be a python interpreter.

   Too bad import doesn't accept a debug flag that prints its search path - but

   I'm guessing for that to work, import would have to become a function, like print.

 bob s.

dhalbert, 2022-12-16:
Sorry, I just meant you could do `ls -R /media/yourname/CIRCUITPY` on the host computer (assuming it was Linux).

I think you would find the same issue with regular Python. It also uses `sys.path` to find things. Its `sys.path` does not include `/lib`, of course, but it does include various library locations. If you replaced only some of the files in a particular library, I think it would have the same import troubles.

schaefer01, 2022-12-16:
interesting automated formatting in the issues tool, right angle brackets seem to be converted into shadow indent pipes

dhalbert, 2022-12-16:
Use the `< >` to convert text you paste into monospaced literal text. It will surround the text with triple or single backquotes. I have edited your post above to do that. Click on `Edit` in the `...` menu to see what I did.

Try `ls -R /lib` to get a tree view of all of `lib`.

schaefer01, 2022-12-16:
eventually I'll get there, need caffeine to enervate the grey matter

% ls -R ./lib/ | grep wiznet
./lib//adafruit_wiznet5k:
adafruit_wiznet5k.mpy
adafruit_wiznet5k_dhcp.mpy
adafruit_wiznet5k_dns.mpy
adafruit_wiznet5k_ntp.mpy
adafruit_wiznet5k_socket.mpy
adafruit_wiznet5k_wsgiserver.mpy

dhalbert, 2022-12-16:
The `wiznet/` directory makes the collection of files be a package. Did you create a `CIRCUITPY/wiznet/` directory and then put the `.py` files inside it, or did you copy the `.py` files directly to the CIRCUITPY? If the latter, that would explain the issue you had.

schaefer01, 2022-12-16:
The latter.
Thank you for providing insight.
I'm done, going to self-close.

closed 2022-12-16 by schaefer01

RetiredWizard, 2022-12-16:
@schaefer01 Shameless plug incoming :grin: if you grab PyDOS.py from  https://github.com/RetiredWizard/PyDOS/blob/main/PyDOS.py (you only need the one file) and type import PyDOS you can then run some DOS type commands including dir/s or dir/w/s for a recursive directory listing....