adafruit/circuitpython #7352: any limitation pulseio.PulseIn maxlen and pulse length on M5Stack Atom ?

musimasami, 2022-12-16:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.5 on 2022-12-08; M5Stack Atom Lite with ESP32
Adafruit CircuitPython 8.0.0-beta.0 on 2022-08-18; PCA10059 nRF52840 Dongle with nRF52840
```

### Code/REPL

```python
import adafruit_irremote
import pulseio
decoder = adafruit_irremote.GenericDecode()

pulsein = pulseio.PulseIn(board.D32, maxlen=600, idle_state=True)
pulses = decoder.read_pulses(pulsein,max_pulse=91000,pulse_window=0.104,blocking_delay=0.102)
```

### Behavior

M5Stack Atom
pulse count: 128 Max: 9140usec

PCA10059
pulse count: 167 Max: 65535usec

### Description

I tested IR Remote signal recieveing both PCA10059 and M5Stack Atom Lite.
M5Stack Atom Lite can't detect long pulse.
Show same signal result below. (83 pulses) + (65535usec space) + (83 pulses)

Adafruit CircuitPython 8.0.0-beta.0 on 2022-08-18; PCA10059 nRF52840 Dongle with nRF52840

import adafruit_irremote
import pulseio
decoder = adafruit_irremote.GenericDecode()

pulsein = pulseio.PulseIn(board.P0_31, maxlen=600, idle_state=True)
pulses = decoder.read_pulses(pulsein,max_pulse=91000,pulse_window=0.104,blocking_delay=0.102)
Set maximum pulse length 91000usec.
pulses sum: 174568 pulse count: 167 Max: 65535usec 
pulses: 
[3489, 1722, 459, 440, 433, 439, 433, 1285, 460, 1287, 457, 440, 433, 1285, 459, 440, 433, 439, 432, 441, 432, 1286, 458, 441, 432, 440, 433, 1285, 459, 441, 432, 1286, 459, 440, 433, 1285, 458, 441, 433, 439, 432, 1286, 459, 441, 431, 440, 433, 440, 432, 440, 432, 1287, 458, 1287, 458, 440, 433, 440, 432, 1287, 457, 1286, 459, 440, 433, 440, 432, 440, 432, 1287, 458, 440, 432, 1287, 458, 1286, 458, 1287, 459, 440, 432, 440, 433, 65535, 3476, 1722, 460, 439, 433, 440, 433, 1285, 459, 1286, 458, 440, 434, 1284, 459, 440, 433, 440, 432, 441, 433, 1285, 458, 441, 433, 439, 432, 1288, 457, 440, 432, 1287, 458, 440, 433, 1287, 457, 440, 433, 439, 433, 1287, 458, 440, 433, 439, 432, 441, 432, 440, 433, 1286, 458, 1286, 459, 440, 433, 439, 433, 1287, 457, 1286, 459, 440, 432, 441, 432, 440, 432, 1286, 458, 441, 432, 1286, 460, 1285, 459, 1285, 459, 440, 433, 440, 432]

Adafruit CircuitPython 8.0.0-beta.5 on 2022-12-08; M5Stack Atom Lite with ESP32

import adafruit_irremote
import pulseio
decoder = adafruit_irremote.GenericDecode()

pulsein = pulseio.PulseIn(board.D32, maxlen=600, idle_state=True)
pulses = decoder.read_pulses(pulsein,max_pulse=91000,pulse_window=0.104,blocking_delay=0.102)

pulses sum: 94214 pulse count: 128 Max: 9140usec 
pulses: 
[3513, 1713, 465, 408, 465, 411, 462, 1278, 468, 1278, 465, 411, 462, 1278, 468, 408, 462, 411, 462, 408, 465, 1278, 468, 408, 465, 432, 438, 1281, 465, 432, 441, 1278, 465, 435, 438, 1278, 468, 432, 441, 432, 438, 1278, 468, 405, 465, 435, 438, 435, 435, 438, 438, 1278, 468, 1275, 468, 435, 438, 432, 441, 1278, 465, 1278, 468, 405, 468, 432, 441, 405, 465, 1278, 468, 432, 441, 1275, 468, 1278, 468, 1275, 468, 408, 465, 432, 441, 9140, 3510, 1716, 468, 432, 441, 432, 435, 1281, 465, 1281, 465, 408, 465, 1278, 465, 423, 453, 405, 465, 411, 462, 1278, 465, 411, 462, 411, 462, 1278, 465, 408, 465, 1278, 465, 411, 462, 1281, 465, 411, 462, 408, 462, 1281, 465, 411]

tannewt, 2022-12-16:
Looks like there is an IDF ringbuffer that we may not be pulling from fast enough: https://github.com/adafruit/circuitpython/blob/2e93c9577a0da46930934fdb632d7faff0e6f57a/ports/espressif/common-hal/pulseio/PulseIn.c#L119

musimasami, 2022-12-17:
The maximum ir signal I'm using is 222 pulses. Panasonic air conditioner.
M5Stack Atom is ESP32-PICO chip.
How about ESP32-C3 & ESP32-S3 & ESP32-S2 chip.

musimasami, 2022-12-18:
**Correction of information**
I measured Panasonic air conditioner pulses. I show it below.

Adafruit CircuitPython 8.0.0-beta.0 on 2022-08-18; Raspberry Pi Pico with rp2040pulses sum: 249428 pulse **count: 439** Max: 10480
 pulses: 
[3521, 1731, 453, 419, 455, 1310, 453, 420, 453, 420, 454, 420, 453, 420, 454, 420, 452, 422, 453, 421, 453, 420, 453, 421, 453, 420, 452, 422, 452, 1312, 451, 422, 451, 423, 450, 423, 450, 423, 440, 433, 433, 441, 448, 425, 431, 1333, 428, 1337, 427, 1336, 428, 445, 428, 445, 428, 1337, 427, 446, 428, 445, 428, 445, 428, 445, 428, 446, 428, 446, 427, 446, 428, 446, 428, 446, 428, 445, 428, 445, 429, 445, 428, 446, 428, 446, 428, 445, 428, 446, 428, 445, 428, 446, 428, 446, 427, 446, 428, 446, 428, 445, 429, 446, 428, 445, 428, 446, 428, 445, 428, 446, 428, 446, 427, 446, 428, 446, 428, 1336, 427, 1337, 427, 446, 427, 446, 428, 446, 427, 446, 428, 445, 428, 10480, 3469, 1764, 427, 450, 427, 1341, 427, 451, 428, 450, 428, 450, 428, 450, 428, 449, 428, 450, 428, 450, 427, 450, 428, 450, 428, 450, 428, 450, 428, 1340, 427, 450, 428, 449, 428, 450, 428, 450, 428, 451, 427, 450, 428, 450, 428, 1341, 427, 1341, 427, 1340, 427, 450, 428, 451, 427, 1341, 427, 450, 428, 450, 428, 450, 428, 450, 428, 451, 427, 450, 428, 451, 427, 450, 428, 450, 428, 450, 428, 450, 427, 451, 428, 451, 427, 1341, 427, 450, 427, 450, 429, 1340, 427, 451, 427, 450, 428, 1341, 427, 450, 427, 450, 428, 1341, 427, 450, 428, 1341, 427, 450, 428, 1340, 427, 450, 428, 451, 428, 449, 428, 450, 427, 450, 429, 450, 427, 450, 428, 450, 428, 450, 428, 1341, 427, 450, 428, 450, 428, 1340, 427, 450, 428, 1341, 427, 449, 428, 1341, 427, 451, 427, 1341, 428, 449, 428, 1341, 427, 1340, 428, 449, 428, 450, 428, 450, 428, 449, 428, 450, 428, 450, 428, 450, 427, 450, 428, 450, 428, 450, 428, 451, 427, 450, 428, 450, 428, 1340, 427, 1340, 427, 1341, 428, 450, 428, 450, 428, 450, 428, 449, 428, 450, 428, 450, 428, 450, 427, 450, 428, 451, 428, 1340, 427, 1342, 426, 1341, 428, 449, 428, 451, 427, 450, 428, 450, 428, 450, 428, 450, 427, 450, 428, 450, 428, 449, 428, 450, 428, 450, 428, 450, 428, 450, 428, 449, 428, 450, 428, 450, 428, 450, 427, 1341, 427, 451, 428, 1340, 428, 449, 428, 450, 427, 451, 428, 1341, 426, 451, 428, 450, 428, 450, 428, 450, 427, 451, 428, 450, 428, 1341, 427, 1341, 427, 451, 428, 450, 427, 1341, 427, 449, 428, 450, 428, 449, 428, 450, 428, 1340, 427, 451, 427, 1341, 427, 1341, 427, 450, 428, 1340, 427, 450, 428, 449, 428, 450, 429]

tannewt, 2022-12-19:
> How about ESP32-C3 & ESP32-S3 & ESP32-S2 chip.

They all use the code I cited.

DavePutz, 2022-12-21:
I tested on an ESP32-S2 and was able to get a max pulsein reading of 65262. It was an accurate reading, based on measurements by a logic analyzer. I will see if I can get a M5Stack Atom to test with.

DavePutz, 2022-12-22:
Tested the example code with an ESP32 ( Adafruit Feather HUZZAH32) and was able to consistently handle pulses > 65K.

closed 2022-12-25 by dhalbert

musimasami, 2022-12-25:
I will check new version.

musimasami, 2022-12-25:
I changed Circuitpython version as below.
Adafruit CircuitPython 8.0.0-beta.6-4-g0d3b7770d on 2022-12-25; M5Stack Atom Lite with ESP32

Max pulse length 65535usec. Works for what I use.
But, Pulse counts are too few. 
Pulse count is same as last time (128) .  
pulses sum: 150552 pulse count: 128 Max: 65535 
pulses: 
[3486, 1716, 465, 408, 465, 408, 468, 1275, 468, 1278, 465, 408, 465, 1278, 468, 405, 465, 408, 465, 408, 465, 1278, 468, 408, 462, 408, 465, 1278, 465, 408, 465, 1278, 468, 405, 468, 1278, 465, 405, 468, 408, 465, 1278, 465, 408, 465, 408, 465, 405, 468, 405, 465, 1278, 465, 1281, 465, 408, 465, 408, 465, 1278, 468, 1275, 465, 408, 465, 411, 465, 405, 468, 1275, 468, 405, 468, 1275, 468, 1251, 492, 1281, 465, 408, 465, 405, 468, 65535, 3489, 1713, 465, 408, 465, 408, 465, 1278, 465, 1281, 465, 408, 465, 1278, 465, 408, 465, 405, 468, 408, 465, 1254, 489, 408, 465, 408, 462, 1281, 465, 408, 465, 1278, 465, 408, 465, 1278, 468, 408, 462, 408, 465, 1278, 468, 405]
[3489, 1713, 465, 408, 465, 408, 465, 1278, 465, 1281, 465, 408, 465, 1278, 465, 408, 465, 405, 468, 408, 465, 1254, 489, 408, 465, 408, 462, 1281, 465, 408, 465, 1278, 465, 408, 465, 1278, 468, 408, 462, 408, 465, 1278, 468, 405]

musimasami, 2022-12-25:
I want a wifi function for analyzing the ir remote signal of the air conditioner, so I'm considering introducing a raspberry pi pico w.

dhalbert, 2022-12-25:
@musimasami Does the fix work to fix your #7352 issue?

musimasami, 2022-12-25:
I will close this issue.
 When I need to read a lot of pulses, I propose a new issue.