adafruit/circuitpython #7376: Ulab Vectorize and np.array slice interaction producing unexpected results

dannystaple, 2022-12-23:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0 on 2022-02-24; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
from ulab import numpy as np

new_poses = np.array([
   [924.881, 311.124, 241.301],
   [1386.19, 520.113, 338.311],
   [40.5783, 94.9179, 193.41],
   [8.8905, 1079.85, 251.301],
   [1171.92, 619.98, 74.1977],
   [629.102, 1447.75, 127.651],
   [575.871, 719.038, 225.301],
   [525.669, 501.781, 42.1525],
   [699.721, 1331.84, 33.3177],
   [189.287, 1133.59, 342.301],
   [178.183, 121.003, 124.247],
   [1039.49, 1340.16, 292.044],
   [173.717, 1006.77, 186.301],
   [193.793, 1225.38, 302.185],
   [886.23, 866.955, 39.7328],
   [450.798, 356.609, 217.301],
   [748.301, 47.7649, 249.386],
   [511.64, 789.662, 32.2231],
   [464.355, 719.037, 164.301],
   [1094.59, 790.142, 197.687]
], dtype=np.float)

constrain_360 = np.vectorize(lambda n: n % 360)
intermediate = new_poses[:,2]
output = constrain_360(intermediate)
output2 = np.array([n % 360 for n in intermediate])
print("With vectorize:", repr(output.tolist()))
print("With list comprehension:", repr(output2.tolist()))
```

### Behavior

I expect the two outputs to be the same. The list comprehension will output the correct thing. The vectorize does something else and gives unexpected output for the constrain_360 vectorized function.

```
With vectorise: [241.301, 306.19, 160.113, 338.311, 40.5783, 94.9179, 193.41, 8.8905, 359.85, 251.301, 91.9199, 259.98, 74.1977, 269.102, 7.75, 127.651, 215.871, 359.038, 225.301, 165.669]
With list comprehension: [241.301, 338.311, 193.41, 251.301, 74.1977, 127.651, 225.301, 42.1525, 33.3177, 342.301, 124.247, 292.044, 186.301, 302.185, 39.7328, 217.301, 249.386, 32.2231, 164.301, 197.687]
```
My testing has shown that if I use a list of dimension 1 (ie, I don't do the multidimensional numpy slice) I cannot reproduce this.

Swapping out the ulab part, and running this on a laptop gives the following (expected) output:
```
With vectorise: [241.301, 338.311, 193.41, 251.301, 74.1977, 127.651, 225.301, 42.1525, 33.3177, 342.301, 124.247, 292.044, 186.301, 302.185, 39.7328, 217.301, 249.386, 32.2231, 164.301, 197.687]
With list comprehension: [241.301, 338.311, 193.41, 251.301, 74.1977, 127.651, 225.301, 42.1525, 33.3177, 342.301, 124.247, 292.044, 186.301, 302.185, 39.7328, 217.301, 249.386, 32.2231, 164.301, 197.687]
```

This suggests it is specific to the ulab implementation.

### Description

- Output becomes corrupted when using ulab.vectorize.
- Only when the input to vectorize is an np.array slice (like `new_poses[:,2]`)
- not reproducible with desktop numpy.

jepler, 2022-12-23:
@v923z I can verify there's something odd going on. I built current ulab for micropython with the `built.sh` script then ran this simplified reproducer:
```py
from ulab import numpy as np

new_poses = np.array([
    [1, 2, 3],
    [4, 5, 6],
    [7, 8, 9],
], dtype=np.float)

identity = np.vectorize(lambda n: n)
intermediate = new_poses[:,2]
output = identity(intermediate)
output2 = np.array([n for n in intermediate])
print("With vectorize:", repr(output.tolist()))
print("With list comprehension:", repr(output2.tolist()))

```
It prints:
```
With vectorize: [3.0, 4.0, 5.0]
With list comprehension: [3.0, 6.0, 9.0]
```

v923z, 2022-12-23:
Thanks for raising the issue. I see that this is a fundamental problem, so I'll try to roll out a speedy fix.

v923z, 2023-01-03:
@dannystaple Fixed in https://github.com/v923z/micropython-ulab/pull/568.

PR #7413 (2023-01-04): update to ulab 6.0.2
Closes #7376
closed 2023-01-04 by dhalbert